suculent/thinx-aes-lib #2: Not correct result for AES128

openact, 2018-07-18:
it is not giving correct result as expected.
I used aes_key[] = { 0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30}
aes_iv[16]= {0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0} 
plaintext[]="AAAAAAAAA"
it should give result as 9D6f45lcvhAK2RAbg7CzvQ==
but giving result as c4OJpl5ibzzBTToos49SsXk8fPWQAQyfxyRJrl3mNAU=
on further debugging I think it is doing wrong padding in method addplaintext. and therefore doing wrong base64 encoding of plaintext.

is someone else getting same wrong result?

suculent, 2018-07-25:
This is just a wrapper. padPlaintext is a dependence and it works for me. Should depend on N_BLOCK size which is recalculated on do_aes_...

suculent, 2018-07-30:
So i've introduced explicit encode/decode and encode64/decode64 functions, that work with/without additional B64 encoding for transport. Let me know if that works for you, please.

openact, 2018-07-31:
> Seems like the issue you've mentioned is here:
> 
> int paddedLen = msgLen + (N_BLOCK - (msgLen % N_BLOCK)) + 1;  // crashes without +1
> 
> It should pad "AAAAAAAAA" (9 bytes) to 16 bytes, but the trailing byte is added because it crashes otherwise.
yes.. from what i could understood.. issue seems to be there..  i could not run base64_iv.ino as it kept crashing..

suculent, 2018-07-31:
Hmm, I'm suspecting prinf commands. The thing is, AES is encrypting data blocks that do not have '\0' C-string termination byte. Therefore printing out those causes a crash. That's one more reason for using Base64.

suculent, 2018-08-13:
Fixed in 1.0.3:
https://github.com/suculent/thinx-aes-lib/commit/8a6a983c7728a9b9e611fe6d7585650428cde770

closed 2018-08-13 by suculent