suculent/thinx-aes-lib #3: Won't compile with Arduino IDE

igormp, 2018-07-26:
When I tried to run the example code found in the examples/readme, I got the following error:

```
Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Uno"

WARNING: Category '"Data Processing"' in library AESLib is not valid. Setting to 'Uncategorized'
WARNING: library AESLib claims to run on [esp8266 architecture(s) and may be incompatible with your current board which runs on esp32] architecture(s).
C:\Users\imp2\Documents\Arduino\libraries\AESLib\examples\simple\simple.ino: In function 'void aes_init()':

C:\Users\imp2\Documents\Arduino\libraries\AESLib\examples\simple\simple.ino:23:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   encrypt("HELLO WORLD!", aes_iv);

                                 ^

C:\Users\imp2\Documents\Arduino\libraries\AESLib\src\AESLib.cpp: In member function 'String AESLib::decrypt(String, byte*, byte*)':

C:\Users\imp2\Documents\Arduino\libraries\AESLib\src\AESLib.cpp:21:10: error: 'class HardwareSerial' has no member named 'printf'

   Serial.printf("Decrypting message %s of length %i \n", msg.c_str(), len);

          ^

C:\Users\imp2\Documents\Arduino\libraries\AESLib\src\AESLib.cpp: In member function 'void AESLib::decrypt(char*, char*, byte*, byte*)':

C:\Users\imp2\Documents\Arduino\libraries\AESLib\src\AESLib.cpp:53:10: error: 'class HardwareSerial' has no member named 'printf'

   Serial.printf("Decrypting message %s of length %i \n", msg, msgLen);

          ^

exit status 1
Error compiling for board Arduino/Genuino Uno.
```

I tried changing the function from `Serial.printf` to `Serial.println` to no avail. Any suggestions?

suculent, 2018-07-26:
What’s your board? Genuino, ESP32 or ESP8266?

Odesláno z iPhonu

igormp, 2018-07-26:
> What’s your board? Genuino, ESP32 or ESP8266?

Genuino Uno, as seen on the log.

suculent, 2018-07-27:
Then you should use library for Arduino like https://github.com/DavyLandman/AESLib instead. 

Odesláno z iPhonu

igormp, 2018-07-27:
So you don't support Arduino boards? I see.
You could make it clear in your README and on the library manager description (that's where I got it from), since the way you've written it makes it sound like arduino boards are supported.

suculent, 2018-07-27:
Fixed in 
https://github.com/suculent/thinx-aes-lib/commit/63d5b3099b0aa6522d25db136ca4f723b91612ac

closed 2018-07-27 by suculent