suculent/thinx-aes-lib #4: esp8266 wdt reset broken

dreamstruct, 2018-10-10:
hi Matej!
Your aeslib is great and simplifies development
But the problem of madness made me encounter
In the test, the device will be inexplicably restarted
Can you take the time to see what the problem is?

>I used ESP8266 esp-01s
>arduino 1.8.7 ide

error msg :
>INPUT:123456
>
>Ciphertext: cXKGqiFPmpOu/w7reWx8CA==
>Cleartext: 123456
>
>INPUT:123456789
>
>ets Jan  8 2013,rst cause:4, boot mode:(3,6)
>
>wdt reset
>load 0x4010f000, len 1384, room 16 
>tail 8
>chksum 0x2d
>csum 0x2d
>v4ceabea9
>ld

source code

> /* Minimalistic example for Readme */
>#include "AESLib.h"
>
>AESLib aesLib;
>
>String plaintext = "AAAAAAA";
>int loopcount = 0;
>
>char cleartext[256];
>char ciphertext[512];
>
>// AES Encryption Key
>byte aes_key[] = { 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30 };
>
>// General initialization vector (you must use your own IV's in production for full security!!!)
>byte aes_iv[N_BLOCK] = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 };
>
>// Generate IV (once)
>void aes_init() {
>  aesLib.gen_iv(aes_iv);
>  // workaround for incorrect B64 functionality on first run...
>  encrypt("HELLO WORLD!", aes_iv);
>}
>
>String encrypt(char * msg, byte iv[]) {
>  int msgLen = strlen(msg);
>  char encrypted[4 * msgLen]; // AHA! needs to be large, 2x is not enough
>  aesLib.encrypt64(msg, encrypted, aes_key, iv);
>  return String(encrypted);
>}
>
>String decrypt(char * msg, byte iv[]) {
>  unsigned long ms = micros();
>  int msgLen = strlen(msg);
>  char decrypted[msgLen]; // half may be enough
>  aesLib.decrypt64(msg, decrypted, aes_key, iv);
>  return String(decrypted);
>}
>
>void setup() {
>  Serial.begin(115200);
>  aes_init();
>}
>
>void loop() {
>
>  loopcount++;
>
>
>  if (Serial.available() > 0)
>  {
>    String ReadBuffer = Serial.readString();
>    Serial.println("input:" + String(ReadBuffer));
>
>    //Serial.print("DEBUG!!!");
>    sprintf(cleartext, "%s", ReadBuffer.c_str());
>
>  // Encrypt
>  byte enc_iv[N_BLOCK] = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 }; // iv_block gets written to, provide own fresh copy...
>  String encrypted = encrypt(cleartext, enc_iv);
>  sprintf(ciphertext, "%s", encrypted.c_str());
>  Serial.print("Ciphertext: ");
>  Serial.println(encrypted);
>
>  // Decrypt
>  byte dec_iv[N_BLOCK] = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 }; // iv_block gets written to, provide own fresh copy...
>  String decrypted = decrypt(ciphertext, dec_iv);
>  Serial.print("Cleartext: ");
>  Serial.println(decrypted);
>  
>  }
>
>
>//delay(500);
>}

suculent, 2018-10-10:
Stack trace would help. This seems like being reset by watchdog timer when stucked in loop for too long. 

Odesláno z iPhonu

suculent, 2018-10-11:
Hello,

I've looked into the issue and also had some weird crashes on the go. However, without changing anything significant in the AESLib, there's new example that works for me at 
https://github.com/suculent/thinx-aes-lib/commit/825629c04bbee96f47bb74caab52b3ecc533cedf

Can you test whether it works for you, please? I've also added heap logging to see how much of your RAM is spent.

Regarding Your code, there may be a memory leak around lines:

```
String ReadBuffer = Serial.readString();
Serial.println("INPUT:" + String(ReadBuffer));
```

I 've used ```Serial.readStringUntil('\n');``` instead, as this will exit upon entering whole cleartext and prevent looping and encryption in the middle of user entry.

Also, you don't need to convert String to String, that just adds memory fragmentation.

```
    String readBuffer = Serial.readStringUntil('\n'); // set 'newline' (LF) in the serial monitor
    Serial.println("INPUT:" + readBuffer);    
    sprintf(cleartext, "%s", readBuffer.c_str()); // must not exceed 255 bytes; may contain a newline at the end
```

dreamstruct, 2018-10-11:
hi Matej!

Thank you for taking the time to check the questions!

Thanks& regards!!!!

In more than 40 hours of testing, no better code could be found

So far, your code is best

However, there is still a bug, which is derived from (`AES.CPP`).

found that there are bugs in 

> void AES::padPlaintext(void* in,byte* out)
> {
>   memcpy(out,in,size); <---------------------------------BUG
>   for (int i = size-pad; i < size; i++){;
>     out[i] = arr_pad[pad - 1];
>   }
> }

You can try the following

First: input long characters (about 150 or so)

Next: input the short character again, for example: 1

Cycle for a period of time, or several times

ESP8266 will be crash

Such questions drive me crazy

dreamstruct, 2018-10-11:
```
1
INPUT:11111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111
msglen = 107
incoming msg: 11111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111
incoming k-size: 4
incoming v-size: 4
- msgLen
- b64data
- b64len
- paddedLen
- padPlaintext
- do_aes_encrypt
- base64_encode
- strcpy
encrypted = hMGvXrYf86E9M39HjFON6tHiDbtjcQk/KhWUAzGQlw8eLYyt+azI4HpnJ512dLdXAXa/pHm/scNUhlTveqXCcXZAEdP4dP01i1pXtu5PSvIqgz0LhRM1l1V6W6eYvSEgynExtmZIFFUdQ0DBl6aWzQTztwj2xdSI8F8uAUb29tMfraoAhLQLsjPRWvPB8nkYAzTiFrmYbZjqIglc20ojVw==
Ciphertext: hMGvXrYf86E9M39HjFON6tHiDbtjcQk/KhWUAzGQlw8eLYyt+azI4HpnJ512dLdXAXa/pHm/scNUhlTveqXCcXZAEdP4dP01i1pXtu5PSvIqgz0LhRM1l1V6W6eYvSEgynExtmZIFFUdQ0DBl6aWzQTztwj2xdSI8F8uAUb29tMfraoAhLQLsjPRWvPB8nkYAzTiFrmYbZjqIglc20ojVw==
free heap: 49208
2
INPUT:1
msglen = 1
incoming msg: 1
incoming k-size: 4
incoming v-size: 4
- msgLen
- b64data
- b64len
- paddedLen
- padPlaintext
- do_aes_encrypt
- base64_encode

Exception (9):
epc1=0x402030ca epc2=0x00000000 epc3=0x00000000 excvaddr=0x784a3456 depc=0x00000000

ctx: cont 
sp: 3ffffc70 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffe10:  0000000c 4010431f 3ffec9b8 40100f22  
3ffffe20:  3020a913 3ffec9b8 3ffec9b8 3ffffec0  
3ffffe30:  3ffffe70 00000001 3fffff50 3ffffef0  
3ffffe40:  3ffffeb0 0000000f 3ffeec80 40202f88  
3ffffe50:  3ffffeb0 3ffeeb38 3ffeec80 402030d1  
3ffffe60:  3ffe89d8 3ffeeb38 3ffeec80 402030d1  
3ffffe70:  3ffe8904 000000fd 3ffeec80 402030fc  
3ffffe80:  3ffe88e0 00000080 3ffeec80 40203141  
3ffffe90:  3ffffeb0 3ffeeb38 3ffeec80 40202b70  
3ffffea0:  00000010 3ffeeb38 3ffeec80 40202b40  
3ffffeb0:  784a3452 6e714138 762b4135 2f2f4236  
3ffffec0:  58355347 3d3d512f ff1fe800 fd572e19  
3ffffed0:  3d3d514d 00000000 3ffeec80 40202f88  
3ffffee0:  3ffe84dc 3ffffef0 3fffff40 3fffff80  
3ffffef0:  00000011 3ffffed0 3ffeec80 4020321b  
3fffff00:  3ffe8910 00000007 3ffeec80 3fffff50  
3fffff10:  3ffffec0 3fffff90 3ffeec80 40202d06  
3fffff20:  3fffdad0 3fffff90 3ffeec80 40202cea  
3fffff30:  3ffeea2c 3fffff80 3ffeec80 40203414  
3fffff40:  3fffdad0 00000000 3ffeec80 3ffeed38  
3fffff50:  f0718247 0fe4a70a ff1fe8af fd572e19  
3fffff60:  00000000 00000000 00000000 00000000  
3fffff70:  3fffdad0 3ffeea2c 3ffeec80 40202e82  
3fffff80:  00000000 00000000 00000000 00000000  
3fffff90:  00000000 00000000 00000000 3ffefa4c  
3fffffa0:  0000000f 00000001 00000001 4020368d  
3fffffb0:  3fffdad0 00000000 3ffeed30 402036f8  
3fffffc0:  feefeffe feefeffe 3ffe850c 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```

suculent, 2018-10-11:
Yes, I've also chosen this code as a base as it is simple enough and well structured. However, there's been some development recently in other projects (see bottom of README), where we could find answers to our questions (especially AES.cpp which is NOT my code at all, see
[https://github.com/spaniakos/AES/]).

I was in situaation where I've been stucked for 6 weeks with memory management. All it required were second exes of a real C++ programmer. I'm more like spread between high and low-level stuff from CX to assembler.

In latest Arduino/ESP8266 framework, it's suggested to use size_t instead of int, this might be an issue to be fixed.

The ```memcpy(out,in,size);``` depends on `size` global variable, which is assigned at 3 places:

* ```void AES::calc_size_n_pad(int p_size)``` – in 2 places:

```
void AES::calc_size_n_pad(int p_size){
  int s_of_p = p_size - 1;
  if ( s_of_p % N_BLOCK == 0){
      size = s_of_p;
  }else{
    size = s_of_p +  (N_BLOCK-(s_of_p % N_BLOCK));
  }
  pad = size - s_of_p;
}
```

* ```void AES::set_size(int sizel)``` – called by do_aes_decrypt where ```size = sizel;```

dreamstruct, 2018-10-11:
My development environment:

device : NodeMCU 1.0 (ESP-12E Module)
esp8266 Community 2.4.0
esp8266 Community 2.4.1
esp8266 Community 2.4.2
All of these SDK versions will crash

suculent, 2018-10-11:
I would need a code with your stacktrace. Can you paste it or make me a Collaborator on your project? Do you have ESP Exception Decoder installed?

To me, only 2.5.0-dev SDK crashes on network transport.

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 19 results
0x402030ca: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char) at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/HardwareSerial.cpp line 133
0x4010431f: lmacProcessAckTimeout at ?? line ?
0x40100f22: ppEnqueueRxq at ?? line ?
0x40202f88: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/HardwareSerial.cpp line 133
0x402030d1: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char) at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/HardwareSerial.cpp line 133
0x402030d1: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char) at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/HardwareSerial.cpp line 133
0x402030fc: _GLOBAL__sub_I__ZN14HardwareSerialC2Ei at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/HardwareSerial.cpp line 139
0x40203141: Print::print(char const*) at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Print.cpp line 114
0x40202b70: AES::calc_size_n_pad(int) at /Users/sychram/Documents/Arduino/libraries/AESLib/src/AES.cpp line 461
0x40202b40: AES::cbc_encrypt(unsigned char*, unsigned char*, int, unsigned char*) at /Users/sychram/Documents/Arduino/libraries/AESLib/src/AES.cpp line 461
0x40202f88: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/HardwareSerial.cpp line 133
0x4020321b: Print::printNumber(unsigned long, unsigned char) at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Print.cpp line 114
0x40202d06: AESLib::encrypt64(char*, char*, unsigned char*, unsigned char*) at /Users/sychram/Documents/Arduino/libraries/AESLib/src/AESLib.cpp line 7
0x40202cea: AESLib::encrypt64(char*, char*, unsigned char*, unsigned char*) at /Users/sychram/Documents/Arduino/libraries/AESLib/src/AESLib.cpp line 7
0x40203414: UpdaterClass::UpdaterClass() at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Updater.cpp line 421
0x40202e82: base64_encode(char*, char*, int) at /Users/sychram/Documents/Arduino/libraries/AESLib/src/base64.cpp line 115
0x4020368d: String::concat(char const*, unsigned int) at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 433
0x402036f8: String::concat(__FlashStringHelper const*) at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 433
0x40100739: _umm_malloc at /Users/sychram/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1450
```

suculent, 2018-10-11:
Note: I'm doing this:

```
int paddedLen = b64len + (N_BLOCK - (b64len % N_BLOCK)) + 1;
```

Maybe the +1 is missing in ```calc_size_n_pad``` where ```p_size - 1``` seems to cut the '\0' C-string end stop.

dreamstruct, 2018-10-11:
It was installed yesterday and cannot be used normally. Now I will study it
I'm studying to study your lib, not a commercial project. If you have a good idea, I'd like to join
I'm using PHP as a server to decode it

suculent, 2018-10-11:
Your code seems to crash before AESLib.cpp, as the ```- strcpy``` is not in your log:

```
  strcpy(output, (char*)out2);
#ifdef AES_DEBUG
  Serial.println("- strcpy"); 
#endif
```

Also, the ```char out2[4*b64len];``` does allocate a lot. I'd try to change it to ```2*b64len```. Unfortunately I cannot test on the hardware now, maybe later in the night.

suculent, 2018-10-11:
Thanks, but we don't use PHP anymore.

And this is not my lib, it's only a wrapper to standard AES for ESP8266/Arduino (see history in readme).

I need this inside of http://thinx.cloud, if you're interested in something of a scale.

dreamstruct, 2018-10-11:
I'm not good at arduino,
PHP, javascript, rad sutdio (PASCAL), is my main skill

suculent, 2018-10-11:
Me neither, but I'm playing with it over 2 years actively now.

suculent, 2018-10-11:
I'm meanwhile contributing to Arduino/ESP8266 framework to include I2C slave mode support these days. AES and SHA will become important soon, as the I2C is far too hackable.

dreamstruct, 2018-10-11:
installing .....

suculent, 2018-10-11:
Sorry, I'm deleting off-topic messages. We should use another channel for one-to-one messages (Hangouts, Slack or whatever you prefer and is available both on Mobile and Desktop).

dreamstruct, 2018-10-11:
My English is not good, please don't mind my writing very slowly
Don't worry, it's right to delete irrelevant content

suculent, 2018-10-11:
My english is not a first language as well. Where are you from?

As I see, if the ```int msgLen = strlen(msg);``` would use ```sizeof(msg)``` instead, if would already include the +1 byte for '\0' c-string end stop resolving all future issues.

suculent, 2018-10-11:
Also the ```byte cipher[2*b64len];``` could be insufficient for long payloads (3* or 4* would solve it but keep in mind how much stack it takes).

dreamstruct, 2018-10-11:
My country is very backward
The whole world would be hostile to him
Of course you won't be interested
haha

suculent, 2018-10-11:
You mean you're from N.K.?

suculent, 2018-10-11:
I was already experienced couple attack attempts, social engineering type maybe wasn't so expected, especially from newly created GitHub account. :o)

dreamstruct, 2018-10-11:
CHINA

dreamstruct, 2018-10-11:
This country is a joke

suculent, 2018-10-11:
I have good friends in China I work with, especially in the hardware field. I'm ordering from China a lot these days. Hopefully the quality will get more sustainable soon.

suculent, 2018-10-11:
I'm from Czech Republic. This country is a joke as well, in a bit different way,

suculent, 2018-10-11:
Please find me at Twitter under @igraczech nickname so we can DM freely without you having to sacrifice your safety.

dreamstruct, 2018-10-11:
Your country is beautiful!

suculent, 2018-10-11:
Do you know, that if each of you wanted to visit Prague once in a life, we'd have to build 30 airports?

dreamstruct, 2018-10-11:
Don't tell me you're a beautiful girl! hahah

suculent, 2018-10-11:
No, I'm ugly father of two girls who will soon become beautiful, so watch out 💯 %

suculent, 2018-10-11:
Really, use the Twitter or your government will ride you.

dreamstruct, 2018-10-11:
in my country can't use twitter , even Google, youtube 
Very funny.Ha ha

suculent, 2018-10-11:
Unless you're a secret agent. Have you ever tried VPN? Like Avast HideMyAss or I don't know... I can provide you with an VPN account if you can ```ping dev.thinx.cloud```

dreamstruct, 2018-10-11:
C:\Users\SANS>ping dev.thinx.cloud

正在 Ping dev.thinx.cloud [188.166.14.166] 具有 32 字节的数据:
来自 188.166.14.166 的回复: 字节=32 时间=341ms TTL=51
来自 188.166.14.166 的回复: 字节=32 时间=348ms TTL=51
来自 188.166.14.166 的回复: 字节=32 时间=342ms TTL=51
来自 188.166.14.166 的回复: 字节=32 时间=339ms TTL=51

188.166.14.166 的 Ping 统计信息:
    数据包: 已发送 = 4，已接收 = 4，丢失 = 0 (0% 丢失)，
往返行程的估计时间(以毫秒为单位):
    最短 = 339ms，最长 = 348ms，平均 = 342ms

suculent, 2018-10-11:
Seems good. What's your platform? Windows or what? You need an OpenVPN client now.

suculent, 2018-10-11:
What's your time? I need to take care of kids (it's 19.00 here) and will be back about 2-3 hours later.

suculent, 2018-10-11:
If you'll be able to install the OpenVPN client, I'll create an account on our VPN server for you, so you can tunnel all your traffic sacurely to Europe and go on from there...

dreamstruct, 2018-10-11:
Take care of the kids first

I can open it
Thank you for your kindness
What's your twitter account?

suculent, 2018-10-11:
```@igraczech```

suculent, 2018-10-11:
But you may be able to use it only when connected to the VPN if you say, you can't Twitter from China. My friend does, but he probably uses the VPN as well. 

I was trying to create a Weibo account, but it has only Chinese instructions. This is what I call Security :o)

dreamstruct, 2018-10-12:
hey Matej!
Are you free now?
It took 20 hours to find the code
He didn't have a crash bug
But decryption will be abnormal

```
#include <AES.h>
#include <Arduino.h>
AES aes ;

byte *key = (unsigned char*)"0123456789010123";

//byte plain[] = "Add NodeAdd NodeAdd NodeAdd NodeAdd Node";

//real iv = iv x2 ex: 01234567 = 0123456701234567
unsigned long long int my_iv = 36753562;

void printf_begin(void)
{
  //JESUS - For reddirect stdout to /dev/ttyGS0 (Serial Monitor port)
  //stdout = freopen("/dev/ttyGS0","w",stdout);
  delay(500);
  printf("redirecting to Serial...");
  
  //JESUS -----------------------------------------------------------
}

void setup ()
{
  Serial.begin (115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }
  printf_begin();
  delay(500);
  printf("\n===testing mode\n") ;

//  otfly_test () ;
//  otfly_test256 () ;
}

void printArray(String name, byte *arr, int length)
{
    Serial.print(name + ": ");
    for (int i = 0; i < length; i++)
    {
        Serial.write(arr[i]);
    }
    Serial.println();
}

int loopcount= 0;
void loop () 
{
  if (Serial.available() > 0) {

    loopcount++; Serial.println(loopcount); // entry counter
    
    String readBuffer = Serial.readStringUntil('\n');
    Serial.println("INPUT:" + readBuffer);

    testAES128(readBuffer);
    }
}

void testAES128 (String msg)
{

  byte plain[200];

  msg.getBytes(plain, sizeof(plain));
  printArray("Plain message", plain, msg.length());
  printf("Plain size: %i\n", msg.length());

  int plainLength = sizeof(plain)-1;  // don't count the trailing /0 of the string !
  int padedLength = plainLength + N_BLOCK - plainLength % N_BLOCK;

  aes.iv_inc();
  byte iv [N_BLOCK] ;
  byte plain_p[padedLength];
  byte cipher [padedLength] ;
  byte check [padedLength] ;
  unsigned long ms = micros ();
  aes.set_IV(my_iv);
  aes.get_IV(iv);
  aes.do_aes_encrypt(plain,plainLength,cipher,key,128,iv);
  Serial.print("Encryption took: ");
  Serial.println(micros() - ms);
  ms = micros ();
  aes.set_IV(my_iv);
  aes.get_IV(iv);
  aes.do_aes_decrypt(cipher,padedLength,check,key,128,iv);
  Serial.print("Decryption took: ");
  Serial.println(micros() - ms);
  printf("\n\nPLAIN :");
  aes.printArray(plain,(bool)true);
  printf("\nCIPHER:");
  aes.printArray(cipher,(bool)false);
  printf("\nCHECK :");
  aes.printArray(check,(bool)true);
  printf("\nIV    :");
  aes.printArray(iv,16);
  printf("\n============================================================\n");
}

```

dreamstruct, 2018-10-12:
I have sent a message to you on twitter. There seems to be something wrong with my account. I don't know why I can't receive the message.
My twitter account looks stolen and prompts me to change my password

dreamstruct, 2018-10-12:
ESP Exception Decoder has been successfully installed, and his version is only available for 1.06

dreamstruct, 2018-10-12:
[1]
Input : 1

31
Plain size: 1
Encryption used time: 1094ms 
Decryption used time: 1336ms 
PLAIN :31000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
CIPHER:f9e3f02e7ba67843371729066c02f2f5b6c9806c967b3bd62b03ce7e75f732532bb21c67ff5403cac8f274cb7fe48427f0b15212fa82d17a715363dedaa5c50cb8483e66b7c35d37ecdf99969521e2964a676dd6dde26f18f206039bd2bc526d41402b2223521ca55edcd9b0053cfc1
CHECK :31000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
IV    :d4142b2223521ca55edcd9b053cfc1

============================================================
[2]
Input : 33333333333333333333333333333333333333333333333333333333333333333333333

33333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333
Plain size: 71
Encryption used time: 963ms 
Decryption used time: 1223ms 
PLAIN :33333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333300000000000000000000000000000000000000000000000000000000
CIPHER:56974e49eec160dfadc90c989b2d8ed551c1fdca0677dfa721d2e0f91cfbbf2f77c81695653397872e7194af8d3230b9bf02c2478f9cdf4354911a66a6563b37f85c22d8d8c9ca0dd2e8488de3faa947a9b004fa95c4a42cf5184b3fb2a528c26869ab2820eb69c4f95ed9a8b151a5d
CHECK :33333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333300000000000000000000000000000000000000000000000000000000
IV    :26869ab282eb69c4f95ed9a8b151a5d

============================================================
[3]
Input : 01

3031
Plain size: 2
Encryption used time: 944ms 
Decryption used time: 1199ms 
PLAIN :30310000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
CIPHER:79e8d610a383d5a01982ca2b3c13debb5177afbd78d8556a596777a078f6e323d37d75bb37e157cab0cc949df1880be9a7e95b2418ff0aa150d54e876b744a8c15631b8e863a236133b810c3843e380c3e2c839479eab1d2ff4d62537f47f6e59163337ce140483a9ec0f135d8d8212
CHECK :30310000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
IV    :59163337ce14483a9ecf135d8d8212

============================================================
HAHAHA

dreamstruct, 2018-10-12:
[1]
Input : 1

31
Plain size: 1
Encryption used time: 1108ms 
Decryption used time: 1336ms 
PLAIN :31000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
CIPHER:f9e3f02e7ba67843371729066c02f2f5b6c9806c967b3bd62b03ce7e75f732532bb21c67ff5403cac8f274cb7fe48427f0b15212fa82d17a715363dedaa5c50cb8483e66b7c35d37ecdf99969521e2964a676dd6dde26f18f206039bd2bc526d41402b2223521ca55edcd9b0053cfc1
CHECK :31000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
IV    :d4142b2223521ca55edcd9b053cfc1

============================================================
[2]
Input : A

41
Plain size: 1
Encryption used time: 952ms 
Decryption used time: 1217ms 
PLAIN :41000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
CIPHER:c62d2fce45e91d5df09cee0e972c58f3d45da60a2c6ce0c9b267a0b542dffb692af2560108336e6856ed43b43397d5bf60546b3bdad115b63d83638613f242b00025aefd5cfe541fdf5f64bd8cb1cbddf05b5fa49cc0d1c0383d80a041241ca957b8a205bdad1ba4c453b5bd64e463d
CHECK :41000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
IV    :957b8a205bdad1ba4c453b5bd64e463d

============================================================
[3]
Input : CCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCC

4343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343001c0fff3f1f00000016000000b0fbfe3fa0fbfe
Plain size: 119
Encryption used time: 963ms 
Decryption used time: 1222ms 
PLAIN :43434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434344343434343434343434343434343434343434343434343434343434343434343434343434343
CIPHER:83b44f8d73dea7c30783aec25475a4b1254d23fe2eaa2edee0606571f560e49d4455c9efd8240637f0423f54cdad2d061615fcfe6cbef7eaa3d90488ed4efe05088d32723d77e487941ab173df6ec7997e464798a07ceee89cfd0c677f4a2a91d50bb1cb31b4ec46146691db6e2e4ff
CHECK :43434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434343434344343434343434343434343434343434343434343434343434343434343434343434343434343
IV    :1d50bb1cb31b4ec46146691db6e2e4ff

============================================================
hey Matej! 
So far I've measured 4096 bytes
no crash

dreamstruct, 2018-10-12:
so cool
```

redirecting to Serial...
Welcome to my World
This is ESP8266 AES test Sketch
It was developed jointly by Dreamstruct.sans and Thinx.suculent

RAM: 44888 Byte | CPU: 80 Mhz | Cycle: 85984700 
Enter text to be encrypted into console (no feedback) and press ENTER (newline):[1]
Input : 1

31
Plain size: 1
Plain max: 800 char
Encryption used time: 5ms 
Decryption used time: 8ms 
PLAIN :310000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
CIPHER:f9e3f02e7ba67843371729066c02f2f5b6c9806c967b3bd62b03ce7e75f732532bb21c67ff5403cac8f274cb7fe48427f0b15212fa82d17a715363dedaa5c50cb8483e66b7c35d37ecdf99969521e2964a676dd6dde26f18f206039bd2bc52649f986834f78a3aee2414bc0d0f3b0356f8a8aa87dad963b6c7233b8b96820654216728be7e8ea6045d7f53699359ff387ee4a6b5f804297dac6c28e69fa41cd19e19748cd79ab2949d3ae36d6a5bcd7241f2708e638960246b174e088a69d425b2ad5e49b6f7d75bfabc8b922894102f8e3f2a3d7635e2cc51d7d298645dabd3f0aca7049bca6befefc37e23b8d17a4e9f3c7a4a48e4afb5875e0e9d6ffa887bea8dd8d5fde038beb1da290e5ac34e357b6d2acd6742aca33f076eda5719569f91ca50188ba4f534d1827bf63347c06c61a4658209a84ca429038d13ce2020bbd96d6c4d890368056b758f70b50ae5f188368f67e96e9ff8b51a473b107c2199843865ad5aa823c8ca9d47e2552b7f4c41082bcd31dab3719baabd499572fbda63dceaf241cca11458b4f95a0f614f4358c3019f6af3a8c70ee438481cd1592b0bb3d2d82745148a9b270f4e6472384b96b163c7c86178515e2f7de4d17c8106236c0e8548c0df68b2a3802bad0dda9a9ef1ed40f81bf0fc67376a32f29c6849fdabef67f4a7e38801260a152b1decc058ad4e80d937d27ed431d9d5b3663670306f122b58f56fa81edbdda1dd3737601c6b64203cc7757560a7e4661527faac4b9f11e05772d229e356b5607f17a4047c064855e9e049b1b968f5c072f5b0fb7b70dd98dce7a44d039ff3c493f010c7aca740a23bf68bc1397de5b60c80b6b358586cd76d36c4ee9a9c8b22adb314bb83a12bf39a06a446bfab98a1d1e16d30bb245c66819d4513cf346c25291621fe6d5b5346280a530a30718aa5dc8262b1e4e6088332520ebb534f80c32552c26d297b42821066c23167bc2618087727daaed9b8185b07cb46a6b4fe550df4aeb4c7d58275d1c81645b9abf3bc581f8cf4a13d6cefd8f51f3d77a87b9b0fb29823d898fddd033581f03b9640368418618a76b21f08a621fba85c146bada880822afdebdc789
CHECK :310000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
IV    :621fba85c146bada88822afdebdc789

============================================================
RAM: 44336 Byte | CPU: 80 Mhz | Cycle: 890144182 
```

suculent, 2018-10-12:
Cool, so I need to update the AES.cpp to latest from what source?

Odesláno z iPhonu

dreamstruct, 2018-10-14:
I just saw your reply, and I will share it with you when debugging no exceptions
Your code repeats the padding
Just remove it

dreamstruct, 2018-10-14:
Give up String and try using byte or char
String can give you nightmares
In the crude Arduino development environment, the crude code can bring numerous crashes to memory
There is no bug in the long and short characters but it will crash if the number of characters exceeds 500
Can you reply to my message on twitter? Why can't I get any message?
My twitter account is: @dreamstruct

dreamstruct, 2018-10-14:
> Don’t take credits for sticking open-source LEGO pieces together. 20 hours of google search is not a skill but failure. Once this will become part of Arduino framework, you’ll understand.

I don 't know according to you' r e saying that,
Even if I rewrite AESLib

originally
I'll share with you when debugging is complete, but aes.cpp still has a lot of bugs
I'm disappointed
I want to delete the code that has been written

If you think google.com can solve a problem in 20 hours, that's wrong

Only beer and ubiquitous cigarette butts
Of course, there must be perseverance

It's not a perfect solution, it just confuses more people, if they have the skills, if they can even solve it, if they don't have it it's torture

So it's not about helping people, and if businesses need to use the library, they'll think we're writing crap

I need to tell you something important

ESP Exception Decoder doesn't solve crash
Because it's a waste of time

suculent, 2018-10-21:
Closing this issue. If you have still problems, provide the stacktrace. If you have a solution, provide a PR. That’s how it works.

closed 2018-10-21 by suculent