suculent/thinx-aes-lib #5: 'AESLib::gen_iv' is not random

m3th0d, 2019-03-31:
`AESLib::gen_iv` doesn't fill IV with random bytes but with `0xFF`:
```cpp
void AESLib::gen_iv(byte  *iv) {
    for (int i = 0 ; i < N_BLOCK ; i++ ) {
        iv[i]= (byte) 0xFF; // getrnd();
    }
}
```
I can see `AESLib::getrnd()` above:
```cpp
uint8_t AESLib::getrnd()
{
   uint8_t really_random = *(volatile uint8_t *)0x3FF20E44;
   return really_random;
}
```
Is it an accident or by design?

suculent, 2019-04-01:
This os by original design, feel free to submit better implementation. 

Odesláno z iPhonu

m3th0d, 2019-04-01:
Well, I'm not a pro in Arduino. But as I can see it's kinda tricky to get _true_ random on this microcontrollers. I decided to use [Crypto](https://www.arduinolibraries.info/libraries/crypto) library by Rhys Weatherley. The library provides [RNG](https://rweather.github.io/arduinolibs/classRNGClass.html) (Random Number Generator) class, which uses [NoiseSource](https://rweather.github.io/arduinolibs/classNoiseSource.html) class to generate random numbers. There are two implementations of NoiseSource in the library - [RingOscillatorNoiseSource](https://rweather.github.io/arduinolibs/classRingOscillatorNoiseSource.html) and more secure but requires additional electric components [TransistorNoiseSource](https://rweather.github.io/arduinolibs/classTransistorNoiseSource.html). You can read more about it in the links I provided if interested in this topic.

Anyway, it's 100% not a good idea to use same `0xFF`-filled IV. I think the best (the fastest) option here is to remove these two methods from `AESLib` and expect users to init IV in their own way. And fix examples: use default [`random()`](https://www.arduino.cc/reference/en/language/functions/random-numbers/random/) from standard Arduino library. My guess is it's weak, but anyway better than nothing (`0xFF`) and gives an idea about how IV-initialization should be implemented. And add warning comment in this case to your examples, something like: _"Do not use standard Arduino random to fill IV in security-sensitive projects"_.

suculent, 2019-04-04:
> Anyway, it's 100% not a good idea to use same `0xFF`-filled IV.

Reason for this was testing in development, because it does what it does... encrypting always with the same result. Changed to `getrnd()`.

closed 2019-04-04 by suculent