suculent/thinx-aes-lib #13: all code examples are missing bits parameter

jonathan-annett, 2019-11-02:
behold the evidence:

https://github.com/suculent/thinx-aes-lib/blob/dcf316e09db4e7d0f3d90f1e7ceacc2fc11a70b5/src/AESLib.h#L28

https://github.com/suculent/thinx-aes-lib/blob/dcf316e09db4e7d0f3d90f1e7ceacc2fc11a70b5/examples/simple/simple.ino#L40

https://github.com/suculent/thinx-aes-lib/blob/dcf316e09db4e7d0f3d90f1e7ceacc2fc11a70b5/examples/complex/complex.ino#L83

jonathan-annett, 2019-11-02:
it does not inspire confidence about a library if the demo code will not even compile.

suculent, 2019-11-03:
Can you post your compile log? There's nothing like `bits` in recent commits...

jonathan-annett, 2019-11-03:
Look at the github repo.
You dont need a compiler

jonathan-annett, 2019-11-03:
i'm using a different lib now. just thought you might like to fix your repo.

suculent, 2019-11-03:
Thanx, fixed.

closed 2019-11-03 by suculent