suculent/thinx-aes-lib #15: Problem with decrypt the message

emilr256, 2019-12-15:
Hello!
I use an AES lib from this source on a Mega2560.

Have loaded an example code from a sketchbook for this library:
`
/* Minimalistic example for Readme */

#include "AESLib.h"

AESLib aesLib;

String plaintext = "24051984";

char cleartext[256];
char ciphertext[512];

// AES Encryption Key
byte aes_key[] = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 1, 2, 3, 4, 5, 6, 7 };

// General initialization vector (you must use your own IV's in production for full security!!!)
byte aes_iv[N_BLOCK] = { 7, 6, 5, 4, 3, 2, 1, 9, 8, 7, 6, 5, 4, 3, 2, 1 };

// Generate IV (once)
void aes_init() {
  Serial.println("gen_iv()");
  aesLib.gen_iv(aes_iv);
  // workaround for incorrect B64 functionality on first run...
  Serial.println("encrypt()");
  Serial.println(encrypt(strdup(plaintext.c_str()), aes_iv));
}

String encrypt(char * msg, byte iv[]) { 
  int msgLen = strlen(msg);
  Serial.print("msglen = "); Serial.println(msgLen);
  char encrypted[4 * msgLen]; // AHA! needs to be large, 2x is not enough
  aesLib.encrypt64(msg, encrypted, aes_key, iv);
  Serial.print("encrypted = "); Serial.println(encrypted);
  return String(encrypted);
}

String decrypt(char * msg, byte iv[]) {
  unsigned long ms = micros();
  int msgLen = strlen(msg);
  char decrypted[msgLen]; // half may be enough
  aesLib.decrypt64(msg, decrypted, aes_key, iv);
  return String(decrypted);
}

void setup() {
  Serial.begin(230400);
  while (!Serial); // wait for serial port
  delay(2000);
  Serial.println("aes_init()");
  aes_init();
 
  Serial.println("Enter text to be encrypted into console (no feedback) and press ENTER (newline):");
}

/* non-blocking wait function */
void wait(unsigned long milliseconds) {
  unsigned long timeout = millis() + milliseconds;
  while (millis() < timeout) {
    yield();
  }
}

unsigned long loopcount = 0;

void loop() {
   
  if (Serial.available() > 0) {

    loopcount++; Serial.println(loopcount); // entry counter
   
    String readBuffer = Serial.readStringUntil('\n');
    Serial.println("INPUT:" + readBuffer);   
   
    sprintf(cleartext, "%s", readBuffer.c_str()); // must not exceed 255 bytes; may contain a newline

    // Encrypt
    byte enc_iv[N_BLOCK] = { 7, 6, 5, 4, 3, 2, 1, 9, 8, 7, 6, 5, 4, 3, 2, 1 }; // iv_block gets written to, provide own fresh copy...
    String encrypted = encrypt(cleartext, enc_iv);
    sprintf(ciphertext, "%s", encrypted.c_str());
    Serial.print("Ciphertext: ");
    Serial.println(encrypted); 
  }
}
`
Only what i changed is the: text to encrypt to: 24051984
Key: 1, 2, 3, 4, 5, 6, 7, 8, 9, 1, 2, 3, 4, 5, 6, 7
IV: 7, 6, 5, 4, 3, 2, 1, 9, 8, 7, 6, 5, 4, 3, 2, 1
The rest of the code is the normal example from the sketchbook.
I have try to decode the received base64 in my program that i make, but no success. Also, i tested the encoding/decoding with an [online tool](https://cryptii.com/pipes/W21RJA) to encode . At a same page, i used to crate chain to decode with a revers procedure.
If i use a web based encode i can later decode it on my program, but the Arduino encoded base64 not. Also, i can't decode with an online tool and with my offline tool the Arduino encoded base64.

The arduino code is example provided from the library autor to show an aes-base64 encoding, and i also don't see the fault.
Can somebody show me what is wrong od correct the code, to be useful for sending an base64 encrypted aes ciphr that i can later descrypt with my keys and iv that i provided?
Something is wrong, but i don't known what?

suculent, 2019-12-15:
Hi. Sorry to say that, but this library is not much of supported. I am mostly developing using ESP8266 and there should be already replacement implementation in BearSSL or somewhere else in the core.
With regards to Mega2560, I have one so I could test it... but i'm pretty short on time before Christmas.

suculent, 2019-12-15:
Specifically you don't define what ` but no success.` exactly looks like. Can you provide some log output?
Did you try the node.js example that is included as well?

suculent, 2019-12-15:
I've updated all examples to be buildable again after latest contributions which did not cover that (no CI tests yet).
The log output should look like:
```
00:11:43.860 -> gen_iv()
00:11:43.860 -> encrypt()
00:11:43.860 -> msglen = 12
00:11:43.860 -> incoming msg: HELLO WORLD!
00:11:43.860 -> incoming k-size: 16
00:11:43.860 -> incoming v-size: 16
00:11:43.860 -> - msg >>HELLO WORLD!<<
00:11:43.860 -> - msgLen 12
00:11:43.860 -> - bits :16
00:11:43.860 -> - b64data [16]
00:11:43.860 -> - b64len 16
00:11:43.860 -> - b64data SEVMTE8gV09STEQh
00:11:43.860 -> - b64data 53 45 56 4d 54 45 38 67 56 30 39 53 54 45 51 68 
00:11:43.860 -> - paddedLen 16
00:11:43.860 -> - cipherLen 16
00:11:43.860 -> - padPlaintext 53 45 56 4d 54 45 38 67 56 30 39 53 54 45 51 68 
00:11:43.860 -> - do_aes_encrypt 66 6f 42 8d 8f 4b 1d dc b3 38 eb 30 f2 7e f7 67 
00:11:43.860 -> - base64_encode
00:11:43.860 -> - strcpy
00:11:43.860 -> encrypted = Zm9CjY9LHdyzOOsw8n73Zw==
00:11:43.860 -> Zm9CjY9LHdyzOOsw8n73Zw==
00:11:43.860 -> free heap: 51104
00:11:43.860 -> Enter text to be encrypted into console (no feedback) and press ENTER (newline):
```

suculent, 2019-12-15:
Let me know, if this fixes it, please.

suculent, 2019-12-15:
OK, so now I've tested on Arduino Uno and Mini Pro and there are memory issues... those machines are not powerful enough to do this job.

suculent, 2019-12-15:
BTW, this lib does B64 itself.For me, it currently fails against the example as well. I do recommend you to search for another lib.

emilr256, 2019-12-23:
True, it still not works on Mega2560 (Meduino) and Nano also. Will test on ESP8266 and on ESP32 WROOM to see do it works.
At a moment, after more that two weeks of intensiv searching, i have not find any usable AES/Base64 for Nano or Mega2560 
Still thinking to move toward STM32 and ESP32 (mayber ESP8266) because i need an AES funtion.

suculent, 2020-01-02:
@emilr256 it does not work on Nano/Mega because those do not have enough RAM.

closed 2020-01-02 by suculent

suculent, 2020-01-14:
Just to let you know, latest version is tested on UNO, try the updated simple.ino example.