suculent/thinx-aes-lib #16: Text from nodemcu to nodejs

kchatzatogou, 2019-12-29:
Hello there, 
I would like to mention this is NOT an issue, just a request for assistance!

Initially i would like to thank you for the library and the code you have provided.
You have a nodejs example which is working fine. My question is how i can produce the 
ei6NxsBeWk7hj41eia3S0Od26goTtxHvwO6V27LwSW4= (nodejs string to decrypt from your examples) from c++ (from the examples you have provide)?
In other words i wild like to get the encrypted string and put in nodejs for testing
I will appreciate any kind of assist.
Thx for your time

suculent, 2020-01-02:
```
var plaintext = "HELLO";
var plaintext_b64 = Buffer.from(plaintext).toString('base64');
iv = CryptoJS.enc.Hex.parse(plain_iv); // reset IV back to 0
var ebytes = CryptoJS.AES.encrypt( plaintext_b64, key, { iv: iv } );
var ciphertext = ebytes.toString();
console.log("ciphertext: ", ciphertext);
```

kchatzatogou, 2020-01-02:
Hello there and happy new year,
I probably did not explain well my issue. I would like to get string ei6NxsBeWk7hj41eia3S0Od26goTtxHvwO6V27LwSW4= in arduino (c++) in order to send it in nodejs in order to decrypt it. 
Again thanks for your time

suculent, 2020-01-02:
See examples/complex. 

First you need to set aes_key and enc_iv (must be same on both MCU/Server sides). You can then use convenience method encrypt() or call aes.encrypt() on your own. 

String encrypted = encrypt(cleartext, enc_iv);
sprintf(ciphertext, "%s", encrypted.c_str());

kchatzatogou, 2020-01-02:
Thx for your time.

closed 2020-01-02 by kchatzatogou