suculent/thinx-aes-lib #18: Online Encryption and Device Encryption is Different

prettyfury, 2020-01-16:
I have tried encryption and decryption on ESP8266 and its working, but when trying to verify the encryption through online - https://cryptii.com/pipes/aes-encryption, the encrypted results are different. The last 16 bytes of the Encrypted data is Different, when using a 32 bytes length data. While using the 48 Bytes of Data First 32 Bytes is same as in Online AES 128 Encryption, but the Last 16 Bytes are Different.

suculent, 2020-01-16:
Seems like difference in padding.

suculent, 2020-01-17:
Indeed, it's the padding which is not a function on Cryptii, you need to pad with zero bytes manually to end up with correct result:

[image]

prettyfury, 2020-01-20:
Thanks @suculent for the Information About Padding.
In your new example, I have noticed the line "aesLib.set_paddingmode(paddingMode::Array);"
I tried all the types of paddingMode = Array, Space, Null, Random, ZeroLength, CMS, Bit.
The website cryptii.com uses CMS Padding Mode.

Password 	 : passwordpassword	
Plain Text 	 : HELLO WORLD\r
IV			 : 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15

Password   (HEX) : 70 61 73 73 77 6f 72 64 70 61 73 73 77 6f 72 64
Plain Text (HEX) : 48 45 4c 4c 4f 20 57 4f 52 4c 44 0d
IV (HEX)  : 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f

Padding Mode :Space

Cipher Base 64: 4wr/e3Jy+1odrjvVHM8Gww==
Decrypted : HELLO WORLD
SUCCESS

Padding Mode :Null

Cipher Base 64: 4wr/e3Jy+1odrjvVHM8Gww==
Decrypted : HELLO WORLD
SUCCESS

Padding Mode : Array

Cipher Base 64: 4wr/e3Jy+1odrjvVHM8Gww==
Decrypted : HELLO WORLD
SUCCESS

Padding Mode : Random

Cipher Base 64: 4wr/e3Jy+1odrjvVHM8Gww==
Decrypted : HELLO WORLD
SUCCESS

Padding Mode : ZeroLength

Cipher Base 64: 4wr/e3Jy+1odrjvVHM8Gw4f6ofR2wwi0KtuTXo1ZJ38=
Decrypted : HELLO WORLD
SUCCESS

Padding Mode : CMS

Cipher Base 64: 4wr/e3Jy+1odrjvVHM8Gw3jsvPe4FAvNHP7BYMha6U8=
Decrypted : HELLO WORLD
SUCCESS

Padding Mode : Bit

Cipher Base 64: 4wr/e3Jy+1odrjvVHM8GwwmdxqtVqQbnAC1Phe1oVjA=
Decrypted : HELLO WORLD
SUCCESS

[image]

closed 2020-01-20 by prettyfury

suculent, 2020-01-20:
Thanks for finding this out. I've just updated the library with test that proves recent fixes valid.

suculent, 2020-01-20:
[image]