suculent/thinx-aes-lib #20: Examples not working

pfeerick, 2020-02-17:
Have the included examples been tested since the last two releases? It seems that 2.0.3 and 2.0.4 have broken ESP8266/ESP32 Arduino support for at least the 'Simple' and Medium exampls from this repo, and doesn't work with the Arduino Uno even with version 2.0.2. If support for the AVR platform has actually been dropped, can you drop the mention of 'Tested on: Arduino Uno [ok]' from the readme, as well as dropping it from the platformio library architecture list. 

Errors included:

Mega/Uno
```
/home/pfeerick/Arduino/libraries/AESLib/src/AESLib.h:9:10: fatal error: iomanip: No such file or directory
 #include <iomanip>
          ^~~~~~~~~
```

ESP8266 / ESP32
```
/home/pfeerick/Arduino/libraries/AESLib/src/AESLib.cpp: In member function 'uint16_t AESLib::decrypt64(char*, uint16_t, char*, unsigned char*, int, unsigned char*)':
/home/pfeerick/Arduino/libraries/AESLib/src/AESLib.cpp:221:50: error: 'memset_s' was not declared in this scope
   int er = memset_s( out, sizeof(out), 0, b64len );
```

ESP8266 / ESP32  - Medium example:
```
/home/pfeerick/Arduino/libraries/AESLib/src/AESLib.h:49:14: note: uint16_t AESLib::encrypt64(char*, uint16_t, char*, unsigned char*, int, unsigned char*)
     uint16_t encrypt64(char * input, uint16_t input_length, char * output, byte key[],int bits, byte my_iv[]); // base64 encode, encrypt and base64 encode again; will deprecate
              ^
/home/pfeerick/Arduino/libraries/AESLib/src/AESLib.h:49:14: note:   candidate expects 6 arguments, 5 provided
```

suculent, 2020-02-17:
I’ve noticed just yesterday when building with VSCode/PIO for ESP32. Examples are outdated indeed, since the testing covers lib only. 

I’ll hopefully fix it soon. The <iomanip> is not for Arduino and memset_s is not required, it´s just safer on most systems. 

I have it working on Uno... btw. 

M.

suculent, 2020-02-17:
Thanks for reports... all issues are now fixed IN 2.0.5: AVR compatibility and Simple/Medium examples.

closed 2020-02-17 by suculent