suculent/thinx-aes-lib #24: Decryption does not work

lim-jeongmin, 2020-04-17:
#include "AESLib.h"

AESLib aesLib;

String text = "i am happy";

char cleartext[256];
char ciphertext[512];

// AES Encryption Key
byte key[] = { 0x15, 0x2B, 0x7E, 0x16, 0x28, 0xAE, 0xD2, 0xA6, 0xAB, 0xF7, 0x15, 0x88, 0x09, 0xCF, 0x4F, 0x3C };

// General initialization vector (you must use your own IV's in production for full security!!!)
byte my_iv[N_BLOCK] = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 };

String encrypt(char * msg, byte iv[]) {  
  int msgLen = strlen(msg);
  char encrypted[2 * msgLen];
  aesLib.encrypt64(msg, msgLen, encrypted, key, sizeof(key), iv);  
  return String(encrypted);
}

String decrypt(char * msg, byte iv[]) {
  unsigned long ms = micros();
  int msgLen = strlen(msg);
  char decrypted[msgLen]; // half may be enough
  aesLib.decrypt64(msg, msgLen, decrypted, key, sizeof(key), iv);  
  return String(decrypted);
}

void setup() {
  Serial.begin(115200);
  aesLib.gen_iv(my_iv);
  
  text.toCharArray(cleartext, 256);
  // Encrypt
  String encrypted = encrypt(cleartext,  my_iv);
  Serial.print("Ciphertext: ");
  Serial.println(encrypted);

  // Decrypt
  String decrypted = decrypt(ciphertext, my_iv);  

  if(decrypted.length()!=0){
  Serial.print("Cleartext: ");
  Serial.println(decrypted);  
  }
  else
    Serial.print("Cleartext: "); 
}

void loop() {

}

I did the same as the example, but it is not decrypted
>

chim3y, 2020-05-02:
@lim-jeongmin  i also experienced the same problem. AESlib generated different ciphertext as opposed to the one (https://www.devglan.com/online-tools/aes-encryption-decryption) and hence it decrypts differently.

Please, can you kindly suggest how did you fix the problem?

suculent, 2020-05-02:
See https://github.com/suculent/thinx-aes-lib/issues/18 what padding do you use?

ElMohamed, 2020-05-02:
The array my_iv was overwritten by encryption, so you can not use my_iv  again for decryption. Try this for example: 
#include "AESLib.h"

AESLib aesLib;

String text = "i am happy";

char cleartext[256];
char ciphertext[512];

// AES Encryption Key
byte key[] = { 0x15, 0x2B, 0x7E, 0x16, 0x28, 0xAE, 0xD2, 0xA6, 0xAB, 0xF7, 0x15, 0x88, 0x09, 0xCF, 0x4F, 0x3C };

// General initialization vector (you must use your own IV's in production for full security!!!)
byte my_iv[N_BLOCK] = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 };

String encrypt(char * msg, byte iv[]) {
int msgLen = strlen(msg);
char encrypted[2 * msgLen];
aesLib.encrypt64(msg, msgLen, encrypted, key, sizeof(key), iv);
return String(encrypted);
}

String decrypt(char * msg, byte iv[]) {
unsigned long ms = micros();
int msgLen = strlen(msg);
char decrypted[msgLen]; // half may be enough
aesLib.decrypt64(msg, msgLen, decrypted, key, sizeof(key), iv);
return String(decrypted);
}

void setup() {
Serial.begin(115200);
//aesLib.gen_iv(my_iv);
byte my_iv1[N_BLOCK] = { 0x15, 0x2B, 0x7E, 0x16, 0x28, 0xAE, 0xD2, 0xA6, 0xAB, 0xF7, 0x15, 0x88, 0x09, 0xCF, 0x4F, 0x3C };
text.toCharArray(cleartext, 256);
// Encrypt
String encrypted = encrypt(cleartext, my_iv1);
Serial.print("Ciphertext: ");
Serial.println(encrypted);
sprintf(ciphertext, "%s", encrypted.c_str());

byte my_iv2[N_BLOCK] = { 0x15, 0x2B, 0x7E, 0x16, 0x28, 0xAE, 0xD2, 0xA6, 0xAB, 0xF7, 0x15, 0x88, 0x09, 0xCF, 0x4F, 0x3C };
// Decrypt
String decrypted = decrypt(ciphertext, my_iv2);

if(decrypted.length()!=0){
Serial.print("Cleartext: ");
Serial.println(decrypted);
}
else
Serial.print("Cleartext: ");
}

void loop() {

}

suculent, 2020-05-02:
Of course. The iv gets changed with each block so you need to save the INITIAL value to decode the chain of blocks. 

That’s how block cipher works. 

M.

ElMohamed, 2020-05-02:
Yes, indeed.

chim3y, 2020-05-03:
@suculent thank you so much for your suggestion. That does solve the problem of padding. However, i am still not able to decrypt my encrypted text. Actually, i sending encrypted udp message from python using aes 128 cbc (pycryptodome) to Arduino.  On the Arduino side, i used aeslib with the same iv and key. Aeslib can decrypt the encrypted text by itself. But, it cannot decrypt the encrypted text from python's pycrytodome.  

With the following settings:
plaintext = b'The answer is no'
passkey = b'1111111111111111' ( or { 0x31,0x31,0x31,0x31,0x31,0x31,0x31,0x31,0x31,0x31,0x31,0x31,0x31,0x31, 0x31, 0x31 })
IV = b'0000000000000000' (or { 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0 })

Using (https://www.devglan.com/online-tools/aes-encryption-decryption) it generates cipher text=5sOIGqVJp3FBFkLUtXN+UVQZeKAfWyhWAxUTaXHu8nw= (base64 encoded)

Using pycrytodome also generates the same cipher text= 5sOIGqVJp3FBFkLUtXN+UVQZeKAfWyhWAxUTaXHu8nw= (base64 encoded)

Using aeslab it generates different cipher text=
bbk8Hv+CiZKqcwLziDlkOyrWJpqSwHINJUkAML2bPD8= (base64 encoded in esp32)

Hence, it cannot decrypt the base64 encrypted text from pycryptodome.

I look forward to your suggestion. Have a nice day ahead.

ElMohamed, 2020-05-03:
@chim3y. Do you use AesLib::decrypt64 function if so at the python side you need to do the following: encod64(plaintext) - > encrypt(encoded64plaintex) - > encode64(encryptedEncode64plaintext)

Or you need to use AesLib:encrypt().

chim3y, 2020-05-03:
@ElMohamed  thank you so much..
However it only decrypt partially. 

For the plaintext:
`Cleartext UDP: ⸮⸮⸮K⸮⸮I⸮⸮s no`

Here is my python encryption:
`from Crypto.Cipher import AES
from base64 import b64encode, b64decode
from Crypto.Util.Padding import pad, unpad

BLOCK_SIZE = 16

def encrypt(plaintext, key, IV):
    cipher = AES.new(key, AES.MODE_CBC, IV)
    cipher_bytes = cipher.encrypt(pad(b64encode(plaintext), BLOCK_SIZE))
    return b64encode(cipher_bytes)

def decrypt(ct, passkey, IV):
    cipher = AES.new(passkey, AES.MODE_CBC, IV)
    decrypted = cipher.decrypt(ct)
    return((unpad(decrypted, BLOCK_SIZE)))

plaintext = b'The answer is no' # did I convert these bytes correctly from the sample code?
passkey = b'1111111111111111' #have I converted the passkey from the example code correctly?
IV = b'0000000000000000' # have I converted the IV from the example code correctly?

encrypted_bytes = encrypt(plaintext, passkey, IV)
`
Do you have anymore further recommendation? Thank you so much.

chim3y, 2020-05-03:
@ElMohamed @suculent thank you much for your time and suggestions...it finally worked.

ElMohamed, 2020-05-03:
Goed no problem

closed 2020-05-04 by suculent