suculent/thinx-aes-lib #25: simple.ino example not working

horellana, 2020-04-20:
Tried running the simple.ino example on an arduino nano, but it says that the decryption test failed

[image]

suculent, 2020-05-04:
Hmm, the decrypted_bytes seems like overflow.

suculent, 2020-05-06:
Make sure you have null terminated data string of proper length. This seems like the overflow happens after the “password” encrypted string should terminate but instead decoding goes past that. 
Sorry I’m in time pressure, cannot dig deeper. However I am using this lib in another project without issues so I am confident the internals work (under tests). 
Also this does not make much sense to use on Nano.

maaz-shaikh, 2020-07-27:
Change this 
`unsigned char readBuffer[18] = "username:password";`

To`unsigned char readBuffer[20] = "username:password";`

at initial it will show decrypted failed but later it will decrypt

suculent, 2020-09-13:
Makes sense, closing.

suculent, 2020-09-13:
Just a minute... this won't compile. What about:

`unsigned char readBuffer[19] = "username:password\0";`

suculent, 2020-09-13:
Will be fixed in next release.

closed 2020-09-13 by suculent