suculent/thinx-aes-lib #28: base64_IV example encrypt different at arduino and node.js

castrofernando, 2020-05-11:
Hi,
I'm trying to use the library example base64_IV, but find hard time to make it encrypt/decrypt same value in both platform.
I'm just using the examples provided, then:

AESKey = '2B7E151628AED2A6ABF7158809CF4F3C';
IV = = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 }

Arduino message:
plaintext: Looks like key but it's not me.
Ciphertext: 3gyjVxuUxHITR/GpNuNOOf1BFQwoWwQFcNwndmVw5Bm0KHF2pFtgV4gBobdvnO6G
decrypted: Looks like key but it's not me.

NodeJS message:
plaintext: Looks like key but it's not me.
ciphertext:  j0RFVdlKjYrwx17qzHdt45miHZWoErdyOahKNKM6KTM=
Decrypted message:  Looks like key but it's not me.

Also trying in some online AES converters no luck.

Any idea what could be wrong?
Btw, thanks for sharing!

suculent, 2020-09-13:
It should boil down to OAEP padding, as far as I remember. Also, you should not use the base64 example for starters. Use the simple one.

suculent, 2020-09-14:
There was a bug with padding, causing always using CMS padding instead of Zero padding. Fixed in 2.1.9 as well as in the node.js code.

closed 2020-09-14 by suculent