suculent/thinx-aes-lib #29: AES encrypt decrypt

maaz-shaikh, 2020-07-23:
I am using your library for AES128 bit encryption but was unable to decrypt the same data using online tools 
Hardware used Arduino Mega

suculent, 2020-07-23:
https://github.com/suculent/thinx-aes-lib/issues/24

What padding and encoding did you use? See ticket above.

maaz-shaikh, 2020-07-23:
Okay thank you for your help I'll check tomorrow and will let you know and can you please let me know how to change padding 

I am using the "simple" example provided by you without making any changes

maaz-shaikh, 2020-07-28:
Hey can you please let me know how I can implement PKCS#7 padding in CMS my above issue is been resolved
Thanks and Regards

suculent, 2020-07-28:
Sure. See usage of paddingMode defined in AES.h

maaz-shaikh, 2020-07-28:
Okay thank you!

closed 2020-07-28 by maaz-shaikh