suculent/thinx-aes-lib #30: Padding not working in simple example working only for base64

maaz-shaikh, 2020-07-28:
Hi, I tried padding modes in base64 example and was able to encrypt and decrypt the message but I am unable to change padding modes in simple.ino code
Thanks and Regards
Maaz Shaikh

suculent, 2020-09-13:
How did you change padding modes in B64 version? B64 is just on top of generic version, because some people do want to encrypt generic data. B64 version should encode that data before encrypting to prevent null bytes.

maaz-shaikh, 2020-09-14:
I was using the following line
`aesLib.set_paddingmode(paddingMode::Array);`

tried to Array, CMS, and other padding techniques also.

suculent, 2020-09-14:
My outputs from esp_serial.ino:

```
// aesLib.set_paddingmode(paddingMode::CMS);
INPUT:Hello world.
10:39:01.797 -> encrypted = q7Gx2UcGW7Apux9lPRIZlfAjswitq69jmRa4ItyxcUo=
10:39:01.872 -> Ciphertext: q7Gx2UcGW7Apux9lPRIZlfAjswitq69jmRa4ItyxcUo=
10:39:03.778 -> Cleartext: Hello world.
10:39:03.812 -> SUCCESS

// aesLib.set_paddingmode(paddingMode::Array);
10:40:48.244 -> Enter text to be encrypted into console (no feedback) and press ENTER (newline):
10:40:59.202 -> INPUT:Hello world.
10:40:59.202 -> encrypted = q7Gx2UcGW7Apux9lPRIZlZNuobXSIxSA5OafdTFlMl8=
10:40:59.271 -> Ciphertext: q7Gx2UcGW7Apux9lPRIZlZNuobXSIxSA5OafdTFlMl8=
10:41:01.215 -> Cleartext: Hello world.
```

The B64 version is will probably deprecate soon, however it is a way to encrypt non-string data, if needed.

maaz-shaikh, 2020-09-14:
I'll check again and will let you know soon. Maybe I must have encountered some different issues. will let you know

suculent, 2020-09-14:
There was a bug causing the paddingMode being always set to default (zero). Fixed in 2.1.9.

closed 2020-09-14 by suculent