suculent/thinx-aes-lib #31: Compilation error when adding AESLib to Platformio project

mrdc, 2020-09-13:
Hello,

I'm having a compilation issue when I add AESLib to my project in Platformio:

```
Compiling .pio\build\esp32doit-devkit-v1\lib76b\AESLib\base64.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib1e2\HTTPClient\HTTPClient.cpp.o
Archiving .pio\build\esp32doit-devkit-v1\lib76b\libAESLib.a
\.platformio\packages\framework-arduinoespressif32\libraries\HTTPClient\src\HTTPClient.cpp: In member function 'bool HTTPClient::beginInternal(String, const char*)':
\.platformio\packages\framework-arduinoespressif32\libraries\HTTPClient\src\HTTPClient.cpp:263:32: error: 'base64' has 
not been declared
         _base64Authorization = base64::encode(auth);
                                ^
\.platformio\packages\framework-arduinoespressif32\libraries\HTTPClient\src\HTTPClient.cpp: In member function 'void HTTPClient::setAuthorization(const char*, const char*)':
\.platformio\packages\framework-arduinoespressif32\libraries\HTTPClient\src\HTTPClient.cpp:430:32: error: 'base64' has 
not been declared
         _base64Authorization = base64::encode(auth);
                                ^
*** [.pio\build\esp32doit-devkit-v1\lib1e2\HTTPClient\HTTPClient.cpp.o] Error 1
```

Looks like there is a conflict because of custom base64 lib which AESLib uses and base64 lib in ESP32-Arduino.

suculent, 2020-09-13:
Interesting. I'm seeing the same issue, but am not sure this has anything to do with AESLib. Maybe some namespace collision?

Can be solved by commenting-out the `_base64Authorization = base64::encode(auth);` line in HTTPClient.cpp, if base64 authorization is not used.

Otherwise needs to be examined (AESLib does use own base64 implementation instead of the one in core, it will probably have to be renamed first, later reused the system one).

mrdc, 2020-09-13:
Yes, I’ve also tried to rename it - compilation failed (it was some time ago - don’t remember exactly the error).

suculent, 2020-09-13:
Try this branch https://github.com/suculent/thinx-aes-lib/tree/base64-fix – if this won't help, it's not caused by AESLib, but can be still overridden by commenting out the line from post above.

suculent, 2020-09-13:
BTW, I've found out that my project with same error as yours does not include AESLib at all...

mrdc, 2020-09-14:
> Try this branch https://github.com/suculent/thinx-aes-lib/tree/base64-fix – if this won't help, it's not caused by AESLib, but can be still overridden by commenting out the line from post above.

Thanks, it works! Strange that It hasn’t compiled when I had renamed it.

closed 2020-09-14 by mrdc