suculent/thinx-aes-lib #34: Error with Simple example running on Arduino Nano

datedave, 2020-09-17:
Using an Arduino Nano, with the "simple" example provided, the decrkypt fails (output below) - any ideas?

because the decrypt looks almost there, I wonder if there is some buffer over write somewhere?

readBuffer length: 19
Calling encrypt (string)...
Encrypted length = 44
Encrypted. Decrypting...
44
Calling decrypt...; Decrypted bytes: 32
Decrypted cleartext of length: 32
Decrypted cleartext:
username:passwor⸮s⸮lڶ--⸮⸮n=⸮
Decryption test failed.
---

suculent, 2020-09-17:
This seems more like the aes_iv changes on encryption and forgot being reset to same initial state on decryption...

datedave, 2020-09-17:
Ok, so I think I found the error with you example... but you will need to check as I am no expert!

The call to aesLib.encrypt() writes the cypher data back to the same buffer that ir read the clear data from.
This causes a problem if the cypher data is longer than the clear data. 

So looking at the error above, readbuffer was length 19, but Encrypted was length 44, hence a buffer over write. 

Not being familiar with the maths behind AES encryption, but assuming there is always a risk that the encrypted data could be longer than the clear data, I suggest its better to return a new newly allocated buffer with the encrypted data, rather than ovwe write??

hope that helps

datedave, 2020-09-17:
to check the above I increased the size of the readBuffer to longer that the input string.. 

i.e. unsigned char readBuffer[200] = "username:password";

and got the output 
readBuffer length: 200
Calling encrypt (string)...
Encrypted length = 280
Encrypted. Decrypting...
280
Calling degth = 280
Encrypted. Decrypting...
280
lling decDecrypted bytes: 208
Decrypted cleartext of length: 208
Decrypted cleartext:
username:password
Decrypted correctly.

Although that could have been a fluke.

datedave, 2020-09-17:
Again though it is still returning a length bigger than the input - so assuming you are re-allocating buffer internal to lib?

datedave, 2020-09-17:
Digging a bit deeper.. I think my fix above was a fluke.. I can see now that you return the txt into a differnt global buffer

suculent, 2020-09-17:
Well, I'm trying to solve it but probably will have to back to it later with better focus. 

I'm just refactoring some another crazy project...

datedave, 2020-09-17:
Thanks - much appreciated. I'll have a look around myself, but this is probably way out of my league.

suculent, 2020-09-17:
I don't think so. This is actually a beginner's termination error in C-string. I'm sometimes repeating those even after 20 years...

datedave, 2020-09-17:
If I understand you, you are saying that somewhere you are not adding the null '\0' to terminate the C str correctly - right?

Or are you not allowing enough space at the end of a buffer for a '\0'  somewhere?

irineu, 2020-09-20:
getting the same problem here

suculent, 2021-01-20:
There has been issue with missing aesLib.set_paddingmode(0) in examples.

closed 2021-01-20 by suculent