suculent/thinx-aes-lib #37: Decryption error in "medium example"

mrdc, 2020-09-22:
Hello,

I'm having this issue with aes-lib (2.1.9) when I enter 3 characters: fff, aaa, bbb etc:

```
INPUT:fff
encrypted = MO5BFeJhLongSfatmr7H8w==
Ciphertext: MO5BFeJhLongSfatmr7H8w==
Cleartext: fff���������
FAILURE
```

The same for 6(9...etc) characters:

```
INPUT:bbbbbb
encrypted = HGiY8jseEhOy3NqNhTxmww==
Ciphertext: HGiY8jseEhOy3NqNhTxmww==
Cleartext: bbbbbb������
FAILURE
```

Padding issue?

suculent, 2021-01-20:
What padding do you use? I've recently updated examples to set padding always explicitly.

suculent, 2021-01-20:
Please attach your key/iv otherwise the padding is not possible to infer from your question.

suculent, 2021-01-24:
Thanks for your report. This was indeed error in versions <2.2.2, where the decrypted length value was incorrect (based on decrypted length before base64 decoding).

closed 2021-01-24 by suculent