suculent/thinx-aes-lib #38: esp_serial.ino example not working

fforoutan, 2020-12-14:
Hi, I ran esp_serial.ino example on ESP8266 and the answers I got are not the same as any online crypto site. I also tested for all padding methods
byte aes_key[] = {0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11};
byte aes_iv[N_BLOCK] = {0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11, 0x11};

21:03:03.849 -> Enter text to be encrypted into console (no feedback) and press ENTER (newline):
21:04:57.590 -> paddingmode: Array
21:04:57.624 -> INPUT:Hello
21:04:57.624 -> encrypted = PxoUZhjnrrKf0TpojMznbA==
21:04:59.599 -> Cleartext: Hello
21:04:59.599 -> SUCCES
21:05:06.957 -> paddingmode: Bit
21:05:06.990 -> INPUT:Hello
21:05:06.990 -> encrypted = QfUDJwWV++xsWxjaJr1bHw==
21:05:08.979 -> Cleartext: Hello
21:05:08.979 -> SUCCES
21:05:12.646 -> paddingmode: ZeroLength
21:05:12.679 -> INPUT:Hello
21:05:12.679 -> encrypted = yxWbv1IPjTL6GXSQa/U+jA==
21:05:14.666 -> Cleartext: Hello
21:05:14.666 -> SUCCES
21:05:17.553 -> paddingmode: Null
21:05:17.553 -> INPUT:Hello
21:05:17.553 -> encrypted = c7AK1VFsj8Jbr4K1BTPtIA==
21:05:19.573 -> Cleartext: Hello
21:05:19.573 -> SUCCES
21:05:24.443 -> paddingmode: Space
21:05:24.443 -> INPUT:Hello
21:05:24.443 -> encrypted = T0hB9dIoecWdfF3DvIuoEw==
21:05:26.458 -> Cleartext: Hello
21:05:26.458 -> SUCCES
21:05:28.599 -> paddingmode: CMS
21:05:28.599 -> INPUT:Hello
21:05:28.599 -> encrypted = xXKlsa0C9Ncrc1BhURySng==
21:05:30.615 -> Cleartext: Hello
21:05:30.615 -> SUCCES

non of encrypted results is valid in any online crypto.
what is the problem?

suculent, 2021-01-20:
Are you resetting aes_iv in each loop or do you keep it iterate?

The CMS (=0) paddingmode is reportedly compatible / tested with https://cryptii.com

suculent, 2021-01-24:
I've tried this against https://cryptii.com with successful result. You're probably missing the last base64 decoding step. 

[image]

suculent, 2021-01-24:
@fforoutan Your case, first result:
[image]

Closing.

closed 2021-01-24 by suculent