suculent/thinx-aes-lib #40: Arduino Zero compatibility

bsc-itcl, 2021-01-12:
Hi, is it planned to make this library Arduino Zero - compatible?

I've tried to do it myself, fixing the issues I found when running the simple example on my MKRZero board, but I'm stuck:

1. AES_config.h:37:14: fatal error: pgmspace.h: No such file or directory -> changed to #include <avr/pgmspace.h>
2. xbase64.cpp:7:10: fatal error: pgmspace.h: No such file or directory -> changed to #include <avr/pgmspace.h>
3. arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\bits\stl_algobase.h:243:56: error: macro "min" passed 3 arguments, but takes just 2 -> **stuck!**

The last issue seems to be this [one ](https://arduinojson.org/v6/error/macro-min-passed-3-arguments-but-takes-just-2/) regarding SAMD21 microcontroller, but I can't fix it for this library.

Maybe someone who sees the "bigger picture" can tell me if this is just a matter of fixing this bug of finding a workaround for this issue, or it's gonna be more complicated.

Thanks and regards.

suculent, 2021-01-20:
The fix (which is in Arduino Core and affects much more than this library) has been merged to https://github.com/arduino/ArduinoCore-samd...

Current state, is that SAMD21 core is not compatible with rest of the world, so best thing I can do is nothing for now.

closed 2021-01-20 by suculent

keyguru, 2021-01-21:
https://stackoverflow.com/questions/3437404/min-and-max-in-c
Maybe it could be fixed manually by creating missing min macro.

vikramsvdd, 2022-11-14:
```
#undef max
#undef min 
```

maybe adding this statement to the file where the error happened will fit through