suculent/thinx-aes-lib #45: Uexpected encoding result (ESP32)

tilchl, 2021-02-09:
When i encrypt with the lib i get different results then with cyberchef, cryptii, cryptojs.
It's still decodable but only in my c code. I suspect I set the key or iv wrong.

With my code i get:
```yaml
key: 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 
enc IV: 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a
Ciphertext: X1rr7fuUJvEmK10s4EaICb94yXZw+XMIw+g5hWhtO6o=
dec IV: 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 
Cleartext: username:password
```
with [cyperchef](https://gchq.github.io/CyberChef/#recipe=AES_Encrypt(%7B'option':'Hex','string':'6a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a'%7D,%7B'option':'Hex','string':'6a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a%206a'%7D,'CBC','Raw','Raw','Ciphertext')To_Base64('A-Za-z0-9%2B/%3D')&input=dXNlcm5hbWU6cGFzc3dvcmQ) (already configured)

```yaml
key: 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 
enc IV: 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a 6a
Ciphertext: PmCPqlih/pdH2q0OBxh6UQ2MdiAp3OZ2+S+X1XPHTVk=
```
I mostly copied your example
I set `aesLib.set_paddingmode((paddingMode)0);`
Main Code:
````c
  char cleartext[256]="username:password";
  char ciphertext[512];
    Serial.print("key: ");
    for (int i = 0; i < 16; i++) {
      Serial.printf("%x ", aes_key[i]);
    }
    Serial.println();
    byte enc_iv[N_BLOCK]   = { 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A };
    byte dec_iv[N_BLOCK]   = { 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A, 0x6A };
    Serial.print("enc IV: ");
    for (int i = 0; i < 16; i++) {
      Serial.printf("%x ", enc_iv[i]);
    }
    Serial.print("\n");
    // Encrypt
    uint16_t clen = String(cleartext).length();
    String encrypted = encrypt(cleartext, clen, enc_iv);
    sprintf(ciphertext, "%s", encrypted.c_str());
    Serial.print("Ciphertext: ");
    Serial.println(encrypted);
    Serial.print("dec IV: ");
    for (int i = 0; i < 16; i++) {
      Serial.printf("%x ", dec_iv[i]);
    }
    // Decrypt
    uint16_t dlen = encrypted.length();
    String decrypted = decrypt( ciphertext, dlen, dec_iv);
    Serial.print("\nCleartext: ");
    Serial.println(decrypted);
    Serial.println();
````

functions:
```c
String encrypt(const char * msg, uint16_t msgLen, byte iv[]) {
  int cipherlength = aesLib.get_cipher64_length(msgLen);
  char encrypted[cipherlength];
  aesLib.encrypt64(msg, msgLen, encrypted, aes_key, sizeof(aes_key), iv);
  return String(encrypted);
}

String decrypt(char * msg, uint16_t msgLen, byte iv[]) {
  char decrypted[msgLen];
  aesLib.decrypt64(msg, msgLen, decrypted, aes_key, sizeof(aes_key), iv);
  return String(decrypted);
}
```

suculent, 2021-02-09:
> with cyperchef (already configured)

Don’t use encrypt64/decrypt64. Just encrypt/decrypt. 

Happy coding.

closed 2021-02-09 by tilchl

roysG, 2021-05-16:
How to convert the encrypt to get output of base64?

suculent, 2021-05-17:
Those are two steps.

Normally, decrypt operates on byte-array (not a string as it can contain null-bytes in the middle).

The byte-array can be converted to base64 using int base64_encode(char *output, const char *input, int inputLen)
from included xbase64.cpp.

M.

adriancs2, 2023-07-31:
I have used this library and built a working example. You can have a look a my demo project:
https://github.com/adriancs2/arduino.aes.asp.net

I wrote an article about the step by step implementation of this in my blog:

https://adriancs.com/c-sharp/1081/aes-encrypted-http-request-between-arduino-esp32-and-c-asp-net/

I also shared this article with codeproject:

https://www.codeproject.com/Articles/5365769/AES-Encrypted-Data-Transmission-Between-Arduino-ES

The article also show the coding in C#, but however, if C# is not relevant to you, you can skipped the C# part, only go for the C/C++ Arduino codes.

Here's the simplified version:
https://adriancs.com/arduino/1096/arduino-aes-encryption-128-bits-cbc/