suculent/thinx-aes-lib #46: simple example doesnt work on uno

vonkonow, 2021-02-12:
Hi 
I cant get "simple" example to work correctly on uno (No rush, "medium" example works - I will use that instead)

This is the dump from serial terminal:
readBuffer length: 18
Calling encrypt (string)...
Encrypted length = 44
Encrypted. Decrypting...
44
Calling decrypt...; Decrypted bytes: 44
Decrypted cleartext of length: 44
Decrypted cleartext:
#⸮C⸮⸮䡯:=⸮⸮⸮.⸮HC⸮⸮J⸮⸮t⸮}⸮i⸮
Decryption test failed.
---

vonkonow, 2021-02-12:
Hi after spending hours with both simple and medium I couldnt get either to output the same result as the test vectors in:
https://csrc.nist.gov/CSRC/media/Projects/Cryptographic-Algorithm-Validation-Program/documents/aes/AESAVS.pdf
I eventually used a simpler library that worked on the first try: https://github.com/DavyLandman/AESLib
I suspect that i have missunderstood the implementation...

esmee96, 2021-04-17:
Hello 

I have an issue with the simple example. It does not work on the arduino UNO. 

The first characters are often decrypted correctly, but the last two characters are not. Can someone help me? 

See below my output. 

[image]