suculent/thinx-aes-lib #47: Encrypted Key changed after next attempt

albertlt, 2021-03-20:
Hello. I encrypted text: "4A31D880" with aes key: { 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61, 0x61 }. and IV: {0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30,0x30}. paddingMode::Null.

First encrypt: ZIANFxqhMhVqryJ/QjFW7A==
Second encrypt: Jl+J50IW7L6NniwCw3eZFg==
Third encrypt: tPhroIswnTtFeUd5IyeX/g==

Key and IV never change. If I restart the esp32, the encryption will goes back to first one. The first one can be decrypted correctly, the rest cant. Why is that?

albertlt, 2021-03-20:
Ok, so I put the IV as class variable. It seems that the lib automatically change the key after each encrypt. If I put the IV inside the encrypt function, it doesn't change anymore. Is that expected behaviour?

suculent, 2021-03-20:
The IV changes on each encrypted block. That’s why examples explicitly keep iv aside and use always fresh copy. Otherwise it’s usual to send the IV to the other side as part of encrypted data. 

Hope this helps.

albertlt, 2021-03-20:
How can I control the IV manually? I tried to decrypt using IV generated from another host via TCP socket. For example iv: "DkjjzksDIcBrHPQb". The iv arrived in String format. I break it with the following:

char tmp[16];

for(int i = 0; i < 16; i++){
    if(i == 0){
        strcpy(tmp, iv.substring(i, i+1).c_str());
    }else{
        strcat(tmp, iv.substring(i, i+1).c_str()); 
    }
           
    aes_iv_dec[i] = (byte) tmp[i];
}

Why when I print: Serial.println((char *) aes_iv_dec); The output is: DkjjzksDIcBrHPQbDkjjzksDIcBrHPQb

Basically 2 x "DkjjzksDIcBrHPQb". The byte array size for IV is only 16 right? How come it can hold double the size? The decryption always fail. Before, it was ok if using hard coded IV (same iv on both host and esp32).

albertlt, 2021-03-20:
I tried again with host sending iv key: "0000000000000000" --> it decrypted correctly.. If I do Serial.println((char *) aes_iv_dec);, the result is: 00000000000000000000000000000000

albertlt, 2021-03-20:
nvm, error logic in my app. thank you for the great lib :)

closed 2021-03-20 by albertlt