suculent/thinx-aes-lib #50: Help needed: try to convert node-red code to Arduino

iStitch07, 2021-05-06:
Hello.
I have simple function in node-red with cryptoModule

```
var crypto = global.get('cryptoModule');
var iv = Buffer.from([0x17, 0x99, 0x6d, 0x09, 0x3d, 0x28, 0xdd, 0xb3, 0xba, 0x69, 0x5a, 0x2e, 0x6f, 0x58, 0x56, 0x2e]);
var newmsg = {};
var lastToken = flow.get('XiaomiCurrentToken') || null; // token example: 82tqk2SrdzI4Aijm

if(lastToken) {
    var password = 'XXX938D90F704XXX';
    var cipher = crypto.createCipheriv('aes-128-cbc', password, iv);
    var key = cipher.update(lastToken, "ascii", "hex");
    cipher.final('hex');
}
```

As result I have variable "key" in hex encoded with password

Can I do with this lib same?

suculent, 2021-05-07:
You can, but base64 is safer because it has no 0x00 bytes in the middle so it can be easily converted to string. 

Change hex to base64 in the node-red and use the encrypt64 and decrypt64 methods. 

M.

iStitch07, 2021-05-07:
No, no :) in node-red all work fine. I use this function for xiaomi gateway control. 
Now I try to write same on arduino with your lib and don't know how

suculent, 2021-05-07:
Iterate over a byte array with for-loop and use Serial.print(array[i], ‘HEX’); or sprintf to print hex values to output or string. 

M.

iStitch07, 2021-05-07:
My troubles start from first step ))
Logic are this:
1. Get "password" (const)
2. Get IV (const)
3. Encrypt password by lastToken

```
var crypto = global.get('cryptoModule');
var iv = Buffer.from([0x17, 0x99, 0x6d, 0x09, 0x3d, 0x28, 0xdd, 0xb3, 0xba, 0x69, 0x5a, 0x2e, 0x6f, 0x58, 0x56, 0x2e]);
var newmsg = {};
var lastToken = 'vcT9bEapirfUZNyq';
var password = '777938D90F704E5B';
var cipher = crypto.createCipheriv('aes-128-cbc', password, iv);
var key = cipher.update(lastToken, "ascii", "hex");
cipher.final('hex');

newmsg.payload = key;
//newmsg.payload = lastToken;
return newmsg;
```

Result: 79123097a3c00ef1af46ead911f1b827

Because I null in crypto I can't to understand how I can do this on Arduino

What I already do:

```
AESLib aesLib;
unsigned char cleartext[] = "777938D90F704E5B";
byte aes_iv[N_BLOCK] = { 0x17, 0x99, 0x6d, 0x09, 0x3d, 0x28, 0xdd, 0xb3, 0xba, 0x69, 0x5a, 0x2e, 0x6f, 0x58, 0x56, 0x2e }; // var iv in node-red

void aes_init() {
  aesLib.gen_iv(aes_iv);
  aesLib.set_paddingmode((paddingMode)0);
}

```

but I can't understand what I need to do after

iStitch07, 2021-05-08:
My troubles start from first step ))
Logic are this: 
```
The rule of genarate the "key" is: after user recived 16-byte "token" from "heartbeat", encrypt this string use AES-CBC to generate a 16-byte ciphertext and then converted to 32 bytes of ASCII code string.

For example: user configured 16-byte key "0987654321qwerty" and "token" is "1234567890abcdef", the ciphertext is 0x3E,0xB4,0x3E,0x37,0xC2,0x0A,0xFF,0x4C,0x58,0x72,0xCC,0x0D,0x04,0xD8,0x13,0x14。the ”key” is：”3EB43E37C20AFF4C5872CC0D04D81314”
```

Working code in node-red (CryptoModule)
```
var crypto = global.get('cryptoModule');
var iv = Buffer.from([0x17, 0x99, 0x6d, 0x09, 0x3d, 0x28, 0xdd, 0xb3, 0xba, 0x69, 0x5a, 0x2e, 0x6f, 0x58, 0x56, 0x2e]);
var newmsg = {};
var lastToken = 'vcT9bEapirfUZNyq';
var password = '777938D90F704E5B';
var cipher = crypto.createCipheriv('aes-128-cbc', password, iv);
var key = cipher.update(lastToken, "ascii", "hex");
cipher.final('hex');

newmsg.payload = key;
//newmsg.payload = lastToken;
return newmsg;
```

Result: 79123097a3c00ef1af46ead911f1b827

Now I try to do this at Arduino with you lib
readBuffer[17] = token
aes_key[] =  password converted to HEX
 aes_iv[N_BLOCK] = The AES-CBC 128 initial vector from gateway api doc

```
#include <Arduino.h>

#include "AESLib.h"

#define BAUD 115200

AESLib aesLib;

#define INPUT_BUFFER_LIMIT (128 + 1) // designed for Arduino UNO, not stress-tested anymore (this works with readBuffer[129])

unsigned char cleartext[INPUT_BUFFER_LIMIT] = {0}; // THIS IS INPUT BUFFER (FOR TEXT)
unsigned char ciphertext[2*INPUT_BUFFER_LIMIT] = {0}; // THIS IS OUTPUT BUFFER (FOR BASE64-ENCODED ENCRYPTED DATA)

unsigned char readBuffer[17] = "vcT9bEapirfUZNyq";

// AES Encryption Key (same as in node-js example)
byte aes_key[] = { 0x37, 0x37, 0x37, 0x39, 0x33, 0x38, 0x44, 0x39, 0x30, 0x46, 0x37, 0x30, 0x34, 0x45, 0x35, 0x42 };

// General initialization vector (same as in node-js example) (you must use your own IV's in production for full security!!!)
byte aes_iv[N_BLOCK] = { 0x17, 0x99, 0x6d, 0x09, 0x3d, 0x28, 0xdd, 0xb3, 0xba, 0x69, 0x5a, 0x2e, 0x6f, 0x58, 0x56, 0x2e };

// Generate IV (once)
void aes_init() {
  aesLib.gen_iv(aes_iv);
  aesLib.set_paddingmode((paddingMode)0);
}

uint16_t encrypt_to_ciphertext(char * msg, uint16_t msgLen, byte iv[]) {
  Serial.println("Calling encrypt (string)...");
  // aesLib.get_cipher64_length(msgLen);
  int cipherlength = aesLib.encrypt((byte*)msg, msgLen, (char*)ciphertext, aes_key, sizeof(aes_key), iv);
                   // uint16_t encrypt(byte input[], uint16_t input_length, char * output, byte key[],int bits, byte my_iv[]);
  return cipherlength;
}

uint16_t decrypt_to_cleartext(byte msg[], uint16_t msgLen, byte iv[]) {
  Serial.print("Calling decrypt...; ");
  uint16_t dec_bytes = aesLib.decrypt(msg, msgLen, (char*)cleartext, aes_key, sizeof(aes_key), iv);
[... 79 lines cut ...]
  Serial.println("---");
  delay(10000);
}

```

Result:
```
readBuffer length: 17
Calling encrypt (string)...
Encrypted length = 44
CIPHERTEXT BYTES (in Base64): 
l/pk6co4z+XrCrgn/Gc0ShlA1E8b+Z3m6KpIXGFSPgw=
Encrypted. Decrypting...
44
Lets try see HEX:
97FA64E9CA38CFE5EBAB827FC67344A1940D44F1BF99DE6E8AA485C61523EC000000000000000000
Calling decrypt...; Decrypted bytes: 17
Decrypted cleartext of length: 17
Decrypted cleartext:
vcT9bEapirfUZNyq
Decrypted correctly.
```

I can't understand who I do wrong. May be you can help?

suculent, 2021-05-09:
Sorry but this is beyond my free time resources I can offer. 

It would take at least an hour to get to it and dig through, which I feel hard to find recently. 

Maybe it would help, if you’d concisely specify what exactly does not work for you. 

M.

iStitch07, 2021-05-10:
From device documentation:
```
The rule of genarate the "key" is: after user recived 16-byte "token" from "heartbeat", encrypt this string use AES-CBC to generate a 16-byte ciphertext and then converted to 32 bytes of ASCII code string
```

By default ciphertext how I understand encoded by base64, yes? I try to decode (from your example) and convert to HEX
But result is not 32 bytes string and I can't understand why. And ciphertext is not 16-byte, is it 44 byte

suculent, 2021-05-11:
Ciphertext is not encoded by base64 by default. Ciphertext is just encrypted. There are functions that do encoding/decoding with base64 as well (encode64, decode64) but In your case, base64 should not be used at all. Just encrypt using the encrypt function and then print out the resulting array as "hex" values (not base64).

iStitch07, 2021-05-11:
I try it. Anyway ciphertext length = 44byte, not 16

Code
```
#include <Arduino.h>
#include "AESLib.h"

#define BAUD 115200

AESLib aesLib;

#define INPUT_BUFFER_LIMIT (128 + 1)

unsigned char cleartext[INPUT_BUFFER_LIMIT] = {0};
unsigned char ciphertext[2*INPUT_BUFFER_LIMIT] = {0};

unsigned char readBuffer[17] = "vcT9bEapirfUZNyq";

byte aes_key[] = { 0x37, 0x37, 0x37, 0x39, 0x33, 0x38, 0x44, 0x39, 0x30, 0x46, 0x37, 0x30, 0x34, 0x45, 0x35, 0x42 };
byte aes_iv[N_BLOCK] = { 0x17, 0x99, 0x6d, 0x09, 0x3d, 0x28, 0xdd, 0xb3, 0xba, 0x69, 0x5a, 0x2e, 0x6f, 0x58, 0x56, 0x2e };

void aes_init() {
  aesLib.gen_iv(aes_iv);
  aesLib.set_paddingmode((paddingMode)0);
}

uint16_t encrypt_to_ciphertext(char * msg, uint16_t msgLen, byte iv[]) {
  Serial.println("Calling encrypt (string)...");
  int cipherlength = aesLib.encrypt((byte*)msg, msgLen, (char*)ciphertext, aes_key, sizeof(aes_key), iv);
  return cipherlength;
}

void setup() {
  Serial.begin(BAUD);
  Serial.setTimeout(60000);
  delay(2000);
  aes_init();
}

void wait(unsigned long milliseconds) {
  unsigned long timeout = millis() + milliseconds;
  while (millis() < timeout) {
    yield();
  }
}
unsigned long loopcount = 0;

void loop() {

  Serial.print("readBuffer length: "); Serial.println(sizeof(readBuffer));
  sprintf((char*)cleartext, "%s", readBuffer);

  uint16_t msgLen = sizeof(readBuffer);
  uint16_t encLen = encrypt_to_ciphertext((char*)cleartext, msgLen, aes_iv);
  Serial.print("Encrypted length = "); Serial.println(encLen);

  for (size_t i = 0; i < sizeof(ciphertext); i++)
  {
    Serial.print(ciphertext[i], HEX);
  }

  Serial.println("---");
  delay(20000);
}

```

Result:
```
Encrypted length = 44
594F4B435639515A366F3553582B6934592B696667334F7A496F62645146616B554646442B2F54797662453D0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000---
```

But result with this cleartext, IV and KEY must be: 79123097a3c00ef1af46ead911f1b827

iStitch07, 2021-05-11:
I think I have different, not?

byte aes_key[] = { 0x37, 0x37, 0x37, 0x39, 0x33, 0x38, 0x44, 0x39, 0x30, 0x46, 0x37, 0x30, 0x34, 0x45, 0x35, 0x42 };

byte aes_iv[N_BLOCK] = { 0x17, 0x99, 0x6d, 0x09, 0x3d, 0x28, 0xdd, 0xb3, 0xba, 0x69, 0x5a, 0x2e, 0x6f, 0x58, 0x56, 0x2e };

iStitch07, 2021-05-15:
any way: how I understood AES-CBC 
[image]

Encrypted length for 16 byte cleartext must be 16 byte too, but I receive 24 byte 

```
#include <AESLib.h>
#include <Arduino.h>

#define BAUD 115200

AESLib aesLib;

byte aes_iv[16]    = { 0x17, 0x99, 0x6d, 0x09, 0x3d, 0x28, 0xdd, 0xb3, 0xba, 0x69, 0x5a, 0x2e, 0x6f, 0x58, 0x56, 0x2e };
byte aes_key[16]   = { 0x37, 0x37, 0x37, 0x39, 0x33, 0x38, 0x44, 0x39, 0x30, 0x46, 0x37, 0x30, 0x34, 0x45, 0x35, 0x42  };
byte cleartext[16] = {};
unsigned char readbuffer[] = "vcT9bEapirfUZNyq";
unsigned char encoded[255] = "";

uint16_t encrypt_to_ciphertext(char * msg, uint16_t msgLen, byte iv[]) {
  Serial.println("Calling encrypt (string)...");
  int cipherlength = aesLib.encrypt((byte*)msg, msgLen, (char*)encoded, aes_key, sizeof(aes_key), iv);
  return cipherlength;
}

void setup() {
  Serial.begin(BAUD);
  while(!Serial);
  delay(2000);
  Serial.println("\nBooting...");
}

void loop() {
  for (size_t i = 0; i < sizeof(readbuffer)-1; i++)
  {
    byte b = (byte) readbuffer[i];
    cleartext[i] = b;
  }

  //sprintf((char*)cleartext, "%X", readbuffer);
  uint16_t msgLen = sizeof(cleartext);
  uint16_t encLen = encrypt_to_ciphertext((char*)cleartext, msgLen, aes_iv);
  
  Serial.printf("MsgLen: %d \n", msgLen);
  Serial.printf("EncLen: %d \n", encLen);
  Serial.print("Encoded: ");
  for (size_t i = 0; i < sizeof(encoded); i++)
  {
    Serial.printf("%d ", encoded[i]);
  }
  Serial.println("-----------");

  delay(20000);
}

```

iStitch07, 2021-05-15:
oh. finally I found what I need in enc_iv_to variable ))

closed 2021-05-22 by iStitch07