suculent/thinx-aes-lib #55: No randomness in IV

koshkamau, 2021-08-05:
I'm trying to use this library in my Arduino sketch to encrypt data from a sensor (ESP8266) to server.
I'm just at the beginning of this path so I haven't tested actual encryption yet, but from my debug messages seems that IV is not initialized by random data. I always get  0x2F values.
```
byte aes_iv[N_BLOCK] = { 0xA0, 0xA1, 0xA2, 0xA3, 0xA4, 0xA5, 0xA6, 0xA7, 0xA8, 0xA9, 0xAA, 0xAB, 0xAC, 0xAD, 0xAE, 0xAF };

void somefunc(){
    aes_init(); 
    for(int i=0;i<16;i++) DBG_COMPORT.printf("%02x ", aes_iv[i]);
}

void aes_init() {
  aesLib.gen_iv(aes_iv);
  aesLib.set_paddingmode((paddingMode)0);
}

```

On the DBG_COMPORT I get:

2f 2f 2f 2f 2f 2f 2f 2f 2f 2f 2f 2f 2f 2f 2f 2f`

suculent, 2021-08-05:
IV is not random in examples to ease up debugging. 

Once you have random IV, you need to make sure you have it ready on decryption, which often means transmitting it with encrypted block. 

M.

koshkamau, 2021-08-05:
Thank you, I already understand that (it took me a couple of days though :) ).
I do not use examples AS IS. 
I call aes_init() and right after that I print IV contents to comport as I showed above. 
It always has 2f in each byte.
I also tried to write `#define ESP8266  1`  without success.
It looks like srand/rand doesn't work for me (or I made some stupid mistake among my three lines of code) - I will investigate further today.

suculent, 2021-08-05:
Try aesLib.gen_iv(aes_iv);

koshkamau, 2021-08-06:
The problem is in function `AES::getrandom`

1) it seems strange to me - to call `srand()` before each `rand()`, but I'm no expert.
2) In case of ESP8266 before each `rand()` there is a call `srand ((unsigned int)time(NULL));` which resets the value, generated by rand() to a certain value. It sets this value even if I add a few milliseconds delay between calls.
I don't know why it happens I just did a test:
```
DBG_COMPORT.println();
srand ((unsigned int)time(NULL));
DBG_COMPORT.println(rand());
srand ((unsigned int)time(NULL));
DBG_COMPORT.println(rand());
srand ((unsigned int)time(NULL));
DBG_COMPORT.println(rand());
DBG_COMPORT.println(rand());
DBG_COMPORT.println(rand());
DBG_COMPORT.println(rand());
```
This test prints three identical values and then three different values.
It seems that `srand ((unsigned int)time(NULL));` is faulty and it is called before each rand so it actually prevents random number generator to work.

3) In other branch of `AES::getrandom` before each `rand()` there is a call `srand (millis());`
Though it works correctly, it uses milliseconds to initialize RNG. So if we are requesting some random numbers during one millisecond - we get identical numbers.
So, delay of a few ms should be added between calls to `AES::getrandom` or srand() should be called only once somewhere to initialize RNG.

suculent, 2021-08-06:
Well, this is a bit of problem, as ESP8266 does not provide true secure random number generator and the time after boot is usually same. 

There's just RANDOM_REG32 at  *((volatile int*)0x3FF20E44) but I haven't tried that as the code would not be compatible with Arduino anymore – implemetation would have to be split.

suculent, 2021-09-14:
Any advance in this? I'm picking issues to look into...

suculent, 2021-09-15:
The problem is apparently that `void AESLib::gen_iv(uint8_t * iv)` does not call `getrandom()` at all. For a water leakage sensor, you should use stream cipher instead of block cipher. Something like ChaCha/Poly from https://github.com/kostko/arduino-crypto

hansliss, 2022-09-13:
The whole point of a PRNG is that it provides a new value every time you calculate it. You should seed it before using it the first time, but you should not seed it on every call, since that defeats the whole point of a PRNG. If it's called several times in a row within a millisecond and seeded with millis() every time, all of the values will be the same.
I would suggest that you hand this problem over to the user, and tell them that the PRNG must be seeded, *once*, before using the library.