suculent/thinx-aes-lib #56: NodeJS and Arduino Uno ciphertext outputs are different

digidhamu, 2021-08-13:
I could see different AES ciphertext output between NodeJS and Arduino Lib as below. Hence, I could not decrypt in NodeJS from Arduino Uno sent via Serial port. Also, noted that Arduino Uno output is not as per base64 encoding.

### Example used
[Arduino Uno](https://github.com/suculent/thinx-aes-lib/blob/master/examples/simple/simple.ino)
[Node JS](https://github.com/suculent/thinx-aes-lib/tree/master/nodejs)

### Arduino Uno R3 ciphertext Output
`HJ2ByKbDloQ9YPZBhsuI8lh90,9EaDf93mAOpSwPs=`

### NodeJS ciphertext Output
`HJ2ByKbDloQ9YPZBhsuI8pwQNPFhbUu7k3Vfr+kOsGQ=`

Is there anything I am doing wrong here?

suculent, 2021-08-17:
I'll look at it, once I'll have working Arduino. I have found two Nanos/328P but they don't seem to work in latest Arduino IDE/Big Sur. I have an Uno somewhere, but couldn't find it today.

The NodeJS output is apparently OK, the Arduino block has broken Base64. The question is, what happened there.

digidhamu, 2021-08-17:
> I'll look at it, once I'll have working Arduino. I have found two Nanos/328P but they don't seem to work in latest Arduino IDE/Big Sur. I have an Uno somewhere, but couldn't find it today.
> 
> The NodeJS output is apparently OK, the Arduino block has broken Base64. The question is, what happened there.
> 
> Trying on ESP8266 the `simple.ino` encrypted output is:
> 
> `HJ2ByKbDloQ9YPZBhsuI8lh9dkyy9EaDf93mAOpSwPs=`
> 
> Are you sure there's no baudrate or serial setting mismatch somewhere? This seems like a communication issue, as those B64 strings have same beginnings and ends – therefore must have been calculated from same encrypted data.

I have double checked serial baudrate, both source and target are the same and I have validated from other text too.

suculent, 2021-09-14:
Exactly, my ESP returns the string without `�,` part on the serial monitor. I'll have to test that on Arduino as soon as possible.

suculent, 2021-09-14:
Answer to your question is PADDING.

The NodeJS example does not use CMS padding:

```
var ebytes = CryptoJS.AES.encrypt( message, key, {
    iv: iv,
    mode: CryptoJS.mode.CBC,
    padding: CryptoJS.pad.ZeroPadding
});
``

...while the Simple Example for Arduino has it:

```
enum class paddingMode {
CMS, // <---------------------- = (paddingMode)0
Bit,
ZeroLength,
Null,
Space,
Random,
Array
};
```

closed 2021-09-14 by suculent