suculent/thinx-aes-lib #61: Only some of encrypted message correct

ZJouba, 2021-10-04:
Hi, I'm not sure what I'm doing wrong - hopefully you can help me. I'm trying to encrypt some location data and if I compare the encrypted string to one that I generated on Cyberchef, only some of it matches.

So my payload looks like this:
`{"sn":"123456789012345","lat": 00.000000, "long": 00.000000, "batt": 99}`

And my code, sans my aes key variable `aes_key[]` and my iv variables `aes_iv[16], enc_iv[16] and env_iv_to[16]` which are all the same.
```
char *payload = new char[74](); // = {"sn":"123456789012345","lat": 00.000000, "long": 00.000000, "batt": 99}

unsigned char cleartext[73] = {0};
unsigned char ciphertext[108] = {0};

void aes_init() {
  aesLib.gen_iv(aes_iv);
  aesLib.set_paddingmode((paddingMode)0);
}

uint16_t encrypt(char * msg, uint16_t msgLen, byte iv[]) {
  return aesLib.encrypt((byte*)msg, msgLen, (char*)ciphertext, aes_key, sizeof(aes_key), iv);
}

...
sprintf((char*)cleartext, "%s", payload);    
memcpy(enc_iv, enc_iv_to, sizeof(enc_iv_to));
uint16_t msgLen = strlen(payload);
uint16_t encLen = encrypt((char*)cleartext, msgLen, enc_iv);
Serial.println((char*)ciphertext); //This does not match what I get in Cyberchef
...
```

When we compare the Cyberchef output to the decrypted output:
RE4oy42RMF9Qi5LqDkYNagu29yC+bJeR+U+G8qgirKtxDgtfRDB8nhaqyL+SvxAaBiY6Q1UYSpOj1JgB/euQiwa0ttf97q1ZUCeevpiPPYY=
RE4oy42RMF9Qi5LqDkYNagu29yC+bJeR+U+G8qgirKtxDgtfRDB8nhaqyL+SvxAaBiY6Q1UYSpOj1JgB/euQixUJ9ozCjccY+bDC"0
Hope to hear from you soon
Thanks

suculent, 2021-10-04:
What type of padding do you use on Cyberchef? The zero padding might not be a default there. 

Also, having same key and IV is a security hole. 

M.

ZJouba, 2021-10-04:
Is `uint16_t msgLen = sizeof(payload);` supposed to be 2?

ZJouba, 2021-10-04:
I played around manually with the msgLen parameter and found that 77 gives me a decryptable output i.e.:
`uint16_t encLen = encrypt((char*)cleartext, 77, enc_iv);`
. 
The only strange thing is that the output has some appended `.` but I think that is just the padding? So my issue wat the msgLen parameter. Still not sure what it's supposed to be

suculent, 2022-09-21:
Sorry, you've probably discovered this but msgLen in encrypt should be strlen of encrypted string.
In decrypt it is size of the buffer (e.g. decoded base64).

closed 2022-09-21 by suculent

adriancs2, 2023-07-31:
I have a working sample of Arduino project using this library to encrypt text.

You can visit my github page at:
https://github.com/adriancs2/arduino.aes.asp.net

Article blog post:

1st article (full complete):
https://adriancs.com/c-sharp/1081/aes-encrypted-http-request-between-arduino-esp32-and-c-asp-net/

2nd article (simplified):
https://adriancs.com/arduino/1096/arduino-aes-encryption-128-bits-cbc/