suculent/thinx-aes-lib #62: base64_iv example crashing on esp32

heavyStuff3000, 2022-01-29:
1) AES init... paddingMode::ZeroLength
Encrypting "Looks like key but it's not me." using null-IV with ZeroLength padding
Encrypted(1): 3gyjVxuUxHITR/GpNuNOOf1BFQwoWwQFcNwndmVw5BmDSHjBu975vsNP4shjz5Hc
AES IV: 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
Decrypting "Looks like key but it's not me." using null-IV ZeroLength padding
Cleartext: Looks like key but it's not me.
In first iteration this should work (using untouched dec_iv_B) ^^^

2) AES init... paddingMode::ZeroLength
Encrypting "Looks like key but it's not me." using null-IV with ZeroLength padding
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4010e497  PS      : 0x00060530  A0      : 0x800d2418  A1      : 0x3ffb1cf0  
A2      : 0x00000000  A3      : 0x3ffb1d30  A4      : 0x3f4005e4  A5      : 0x0000009c  
A6      : 0x00000004  A7      : 0x00000004  A8      : 0x00000004  A9      : 0x000000ff  
A10     : 0x00000054  A11     : 0x0000003c  A12     : 0x3ffb8218  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4010e494  LEND    : 0x4010e4c8  LCOUNT  : 0x00000003  

ELF file SHA256: 0000000000000000

Backtrace: 0x4010e497:0x3ffb1cf0 0x400d2415:0x3ffb1d10 0x400d29a1:0x3ffb1d30 0x400d2abd:0x3ffb1d90 0x400d18f4:0x3ffb1e30 0x400d1c1f:0x3ffb1ea0 0x400d1fbc:0x3ffb1f80 0x400d3632:0x3ffb1fb0 0x40086246:0x3ffb1fd0

Rebooting...

heavyStuff3000, 2022-01-29:
Thanks a lot if this can work since this will be an important part of my project since I don't want anyone to send a json file to my server and crash it.. My bet is if my nodejs server have to decrypt it will help with security. I'm using crypto-js on my server. I will make a post with my data and decrypt on nodejs server. I like simple code, and I may be dumb but in the beginning I only wanted aes key. I don't really understand what the iv is? Another password for more security?

heavyStuff3000, 2022-01-29:
Also I know this is a stupid question(yep I'm a real noob) but can this run 24/24 or there is a chance that there is a leak somewhere and it will end up crashing? Silly question I know(or guess)

suculent, 2022-09-21:
Did you try the example in nodejs folder?

closed 2022-09-22 by heavyStuff3000