suculent/thinx-aes-lib #63: "Simple" example fails - abandonned?

Smashcat, 2022-02-10:
The example code fails with decryption failure error messages. I've noticed this never worked in the code this was forked from. Is there a library that can also decrypt AES packets?

suculent, 2022-02-10:
The original low-level AES code used to encrypt/decrypt did always worked as it is covered with tests.

Examples may fail from time to time, I have not much of chance to test/fix them these days.

This project will expectedly deprecate in fabvour of [Arduino Crypto](https://rweather.github.io/arduinolibs/crypto.html)

Smashcat, 2022-02-10:
Ok thanks. It's the simple.ino example that fails decryption. The medium example works - seems like the decrypt() method does not actually decrypt properly. The decrypt64() method does though.

hansliss, 2022-09-12:
The version that gets installed by the Arduino library manager is not the same as here, even though the version number is the same. It does an added Base64 encode even in the encrypt() function, where this shouldn't be done. You will need to manually correct the bug in AESLib.cpp by removing the base64 code and changing the return statement.

closed 2022-09-21 by suculent

adriancs2, 2023-07-31:
Actually, this library: https://rweather.github.io/arduinolibs/crypto.html
is my early attempt to implement the AES encryption. The problem is I have a hard time to run a successful working example with that library, and therefore, I end up using this library (AESLib) in my production.

for anyone that is interested, I shared my code at my github:
https://github.com/adriancs2/arduino.aes.asp.net