suculent/thinx-aes-lib #67: Encrypted output does not match AES test websites

franciscogimeno2000, 2022-12-05:
Hello, first of all thank you for your work.

I have downloaded the latest version 2.3.0 and after removing the redefinition of a variable the examples compile ok on an esp32.

But after trying the example ESP SERIAL
Which gives in the encoded output both the text HELLO WORLD! like the one you can enter via rs232 does not match the decryption done on several AES key test pages like "https://www.devglan.com/online-tools/aes-encryption-decryption"

For example, on that page
if I put HELLO WORLD! with the AES key 0000000000000000 which is the one in the example code as 0x30 than in ASCII in "0"
Returns: W+9C00eM05TwYo2/AGKbBw==
This in BCE format, if we encode 

it as CBC format with additional key IV=00000000000000000 also as the code example results in: j6rUYRiMEiYnz/0deXSsNw==

In the serial output of that program HELLO WORLD! results in: JDlanKOWImRU4/R8z53GJA==

I don't know what can happen.

Thanks for your help.

Fran

suculent, 2022-12-05:
Hello, such issues usually have something to do with padding settings, which needs to be same for both ends (CMS, Array or Null).

See https://github.com/suculent/thinx-aes-lib/issues/30

franciscogimeno2000, 2022-12-05:
Hello, thanks again. 
Well, I've tried all of them and it doesn't work. 
The esp_serial example as it is in the examples folder.

The strange thing is that the first encoding of HELLO WORLD! that it does when esp32 starts is this:
encrypt()
encrypted = JDlanKOWImRU4/R8z53GJA==
JDlanKOWImRU4/R8z53GJA==
Enter text to be encrypted into console (no feedback) and press ENTER (newline):

But then I tell it to decode again HELLO WORLD! and the esp32 sends this:
1
INPUT:HELLO WORLD!
encrypted = ebeGH3VfV++b+ruMfLRM/A==
Ciphertext: ebeGH3VfV++b+ruMfLRM/A==
Cleartext: 5�1s=!��E	n1�QH,�P�?��?�?
FAILURE

Get another encoding different from the first and with a failure notice

franciscogimeno2000, 2022-12-14:
I am using version 2.2.3. from this version up, they don't work.

franciscogimeno2000, 2022-12-14:
1
INPUT:HELLO WORLD!
encrypted = u1PUfN8QeImo1zMbYDhzh3vX15SON2HWKDIZ/kJ1drY=
Ciphertext: u1PUfN8QeImo1zMbYDhzh3vX15SON2HWKDIZ/kJ1drY=
Cleartext: HELLO WORLD!
SUCCES

franciscogimeno2000, 2022-12-14:
Hi, I don't have the solution, I've only been testing the esp_serial example in all versions. Why higher versions don't work I don't know for sure. From the little that I have checked, the higher versions fail in the text encoding part.

closed 2022-12-14 by franciscogimeno2000

adriancs2, 2023-07-31:
I have just published a working example of using this library on Arduino ESP32. you can have a look at my github page:

https://github.com/adriancs2/arduino.aes.asp.net

Here's the simpler version of explanation:
https://adriancs.com/arduino/1096/arduino-aes-encryption-128-bits-cbc/

Here's the full complete explanation of implementation:
https://adriancs.com/c-sharp/1081/aes-encrypted-http-request-between-arduino-esp32-and-c-asp-net/
this article includes C# ASP.NET programming, if C# asp.net is not relevant to you, you may skip the C# part and only look at the Arduino C/C++ section.