suculent/thinx-aes-lib #68: Decrypted output has random characters

d0st3r, 2023-02-09:
When I try to use the ESP_Serial example with the defaults settings, I get this

INPUT:test76784
encrypted = pW3QF/XcpWmAy2ymJ9CHhQ==
Ciphertext: pW3QF/XcpWmAy2ymJ9CHhQ==
Cleartext: test76784H⸮⸮?⸮

FAILURE

The decrypted is correct, but it adds these characters (H⸮⸮?⸮) and I can't get rid of them, idk if it's a problem of b64 decode or what, because if i check the ciphertext on an online tool the decryption doesn't have that characters, hope you can help me.

suculent, 2023-02-13:
I can look at it, the example may be deprecated. What’s your platform/hardware to test against?

d0st3r, 2023-02-13:
I'm testing on a nodemcu 1.0, and using the latest version of the library

suculent, 2023-02-16:
I have updated and verified this example both for Arduino IDE and PlatformIO. Works well on my Wemos D1 Mini Pro.

suculent, 2023-02-17:
Closing this as resolved. Please feel free to reopen once you'll have any issues.

closed 2023-02-17 by suculent

adriancs2, 2023-07-31:
This might be because you didn't include the null character before encryption.

Here's a simple working example of mine using AESLib
https://adriancs.com/arduino/1096/arduino-aes-encryption-128-bits-cbc/