suculent/thinx-aes-lib #69: Don't using with NRF51822

NgHieuPickon, 2023-02-13:
Hi everybody,
I have a project with Generic NRF51 (NRF51822) under the Arduino Core of Sandeepmistry (https://github.com/sandeepmistry/arduino-nRF5)
When I start the "simple" example, the console shows "AES_config.h:37:85: fatal error: pgmspace.h: No such file or directory".
I understand that in the Arduino Core of Sandeepmistry there is no such file. But I think there must be a way to use it.
Please help me!

suculent, 2023-02-13:
It should work, your Arduino core has this file in avr/pgmspace.h:

  #if (defined(__AVR__))
    #include <avr/pgmspace.h>
  #else

Try adding #define __AVR__ at the beginning of this file as a workaround. 

What compiler are you using?

suculent, 2023-02-16:
I have succeeded to build for the Sandeep Mistry's Generic NRF51 after couple minor changes. Please, check out version 2.3.4 if it works for you.

suculent, 2023-04-10:
@NgHieuPickon unless this is not an issue anymore, I'd like to close it.

anhkhoaEmbedded, 2023-05-31:
Hi, I try to compile my arduino sketch using AESlib.h for my Arduino PMC, but when I run compile it said not found <pgmspace.h> . I have no clue about this. Does anyone find some solution this ? I really appreciate.

suculent, 2023-05-31:
Well, this Lib has not yet been used on PMC. Generally, this requires couple ´#if defined()’ locations to be changed, because the pgmspace.h location is not currently defined for this architecture. It needs a bit of digging to find out in compiler output, which of the -D args contains the platform (currently __AVR__, ESP and NRF is supported out of the box).See latest nRF commits for required changes to add platform support. I can help with that, but I’m currently busy and have no such hardware. M. 31. 5.

NgHieuPickon, 2023-06-02:
Hi @suculent, 
I tested version 2.3.6 on NRF51822. It works fine with the "sample" example.
Before you publish version 2.3.4, I used [AES Electronic Code Book](https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v8.x.x/doc/8.0.0/s110/html/a00756.html#details) (ECB) on Arduino IDE and it worked fine.
I haven't compared the efficiency of your library and ECB, but both are usable.
Sincerely thank.

closed 2023-06-02 by suculent