suculent/thinx-aes-lib #70: Ram use problem in Mega2560 with encrypt base64_iv sample

motniemtin, 2023-02-16:
I upload this code to my mega2560, and after a loop, RAM in mega decrease quickly, after that, no free ram and fail decrypt in some loop

suculent, 2023-02-16:
Unfortunately, Mega2650 does not have enough RAM to process the Base64 conversion. That's why there is a `simple` example specially designed for low-memory conditions.

motniemtin, 2023-02-17:
It is not small ram problem in mega2560, i tested your code with base64_iv and i found problem with your AESLib.cpp
char *msgOut = (char*)malloc(msgLen);
int b64len = base64_decode(msgOut, msg, msgLen);

If i add `free(msgOut);` problem solved

suculent, 2023-02-17:
I see,

the malloc was a 3rd party contribution, it's bad idea to use malloc in emedded code.

Thanks for review, I'll release the decrypt64 fix in next version.

Matej

suculent, 2023-02-17:
I've tested your suggestion measuring memory consumption on my ATmega2650 and it seems to work. Releasing as 2.3.5. Thanks again.

closed 2023-02-17 by suculent

motniemtin, 2023-02-18:
I really appreciate your contribution and I thank you very much for this, I'm glad to have there