suculent/thinx-aes-lib #72: Support for Arduino SAMD (MKR) boards

suculent, 2023-04-10:
The version 2.3.6 is adapted for building on these boards, but I don't have one to try with.

adriancs2, 2023-07-31:
I have built a working example, which the code should be able to run on any Arduino module/boards.

Here's my work:
https://github.com/adriancs2/arduino.aes.asp.net

I wrote an article about the step by step implementation of this in my blog:

https://adriancs.com/c-sharp/1081/aes-encrypted-http-request-between-arduino-esp32-and-c-asp-net/

I also shared this article with codeproject:

https://www.codeproject.com/Articles/5365769/AES-Encrypted-Data-Transmission-Between-Arduino-ES

adriancs2, 2023-07-31:
Here, I have just published a simplified version of explanation of how to use this library to do AES encryption locally within the Arduino device. This should work on any Arduino boards. Since it's just pure C/C++, so technically, it should also works on any device that support C++ compiled code device.

Here's the article: https://adriancs.com/arduino/1096/arduino-aes-encryption-128-bits-cbc/