suculent/thinx-aes-lib #74: Example "Simple" does not / can not work.

JaBa1202, 2023-06-02:
Running the example will ALWAYS report success but it is a false success report. Even if you change the line decLen = decrypt(...) to decLen = 0, it will report a successful decrypt though it didn't even decrypt it. Actually, I don't think it could ever work. The library indicates that encrypt and decrypt DO NOT use Base64 encoding but the example "simple" does a base64 encode before calling decrypt.

adriancs2, 2023-07-31:
I have used this library and built a working example. You can have a look a my demo project:
https://github.com/adriancs2/arduino.aes.asp.net

adriancs2, 2023-07-31:
Here, I have just published a simplified version of explanation of how to use this library to do AES encryption locally within the Arduino device. This should work on any Arduino boards. Since it's just pure C/C++, so technically, it should also works on any device that support C++ compiled code device.

Here's the article: https://adriancs.com/arduino/1096/arduino-aes-encryption-128-bits-cbc/